evilsocket/pwnagotchi #43: use websockets instead of polling

evilsocket, 2019-10-02:
currently bettercap's /api/event is polled every second, since bettercap supports websockets we should be using that to receive the events

dangilkerson, 2019-10-02:
I'd like to take a crack at this.

evilsocket, 2019-10-02:
that'd be great! https://www.bettercap.org/modules/core/api.rest/

ws in bettercap never got tested properly, so they might not work 100% correctly ... if that happens, i'll fix it :D

dangilkerson, 2019-10-02:
This is the best way to test it :)

hslatman, 2019-10-04:
@dangilkerson: Did you get a shot at it? 

I've tried the bettercap WS support and it seems to return an HTTP 200 instead of 101 when accessing ws://user:pass@localhost:8081. The usual WS headers are set, though 😕 I used the [websockets](https://github.com/aaugustin/websockets) library to connect with Python.

evilsocket, 2019-10-04:
@hslatman did you set `api.rest.websocket` to true before starting the API module in bettercap?

hslatman, 2019-10-04:
@evilsocket Yes and I've verified it's showing as 'on' in the web interface.

evilsocket, 2019-10-04:
mmm i have no idea ... this is the implementation https://github.com/bettercap/bettercap/blob/master/modules/api_rest/api_rest_ws.go

hslatman, 2019-10-04:
Got it ... 😅 

The parameter does not react to changes that are sent using the API, so I started bettercap with `bettercap -eval 'set api.rest off; set api.rest.websocket true; set api.rest on;'`. Now I've got something going and will try some stuff with it.

Thanks @evilsocket, your first response was right!

evilsocket, 2019-10-04:
oh yeah, in order for parameters to be changed you either set them before the module starts, or restart the module yourself :D glad you figured it out, i'm very looking forward to see what's next @hslatman 👍

dangilkerson, 2019-10-04:
@hslatman Looking at it today but if you've got something going I can move on to something else.

dangilkerson, 2019-10-04:
Watching events over websockets through a test script now. Going to replace the event polling code.

evilsocket, 2019-10-04:
i'm very impressed by the fact this works out of the box XD

dangilkerson, 2019-10-04:
Oh don't worry there's issues I'll probably have changes for bettercap as well ;)

hslatman, 2019-10-04:
> @hslatman Looking at it today but if you've got something going I can move on to something else.

No, I was looking into using the websocket functionality within a different project. I don't have pwnagotchi running, so get your best shot at it ;-)

hslatman, 2020-04-06:
Just found out about this issue again. I've created a small POC repository with an example for using bettercap with WebSockets. It's largely based on the Agent/Client setup in this repository, with some WebSocket specific steps included, so it could be an option to integrate the WebSocket support in this repository at some point and when constraints allow for this.

It's available [here](https://github.com/hslatman/bettercappy.ws).

dadav, 2020-04-07:
Derek assign: dadav

dadav, 2020-04-07:
Hmmm. The connection always gets terminated exactly after 1min and shows this msg:
https://github.com/bettercap/bettercap/blob/4eead7eafaec57282ad435473e420d7bad7a602e/modules/api_rest/api_rest_ws.go#L98

hslatman, 2020-04-07:
> Hmmm. The connection always gets terminated exactly after 1min and shows this msg:
> https://github.com/bettercap/bettercap/blob/4eead7eafaec57282ad435473e420d7bad7a602e/modules/api_rest/api_rest_ws.go#L98

That's not the same behavior as I've observed running my small POC. Perhaps the asyncio task keeping the connection is finished / killed without informing you about it, resulting in no ping being sent at all.

dadav, 2020-04-08:
> 
> 
> > Hmmm. The connection always gets terminated exactly after 1min and shows this msg:
> > https://github.com/bettercap/bettercap/blob/4eead7eafaec57282ad435473e420d7bad7a602e/modules/api_rest/api_rest_ws.go#L98
> 
> That's not the same behavior as I've observed running my small POC. Perhaps the asyncio task keeping the connection is finished / killed without you knowing, resulting in no ping being sent at all.

Maybe thats it! I will switch to `create_task` + `run_forever` instead of `run_until_complete`. Also I should have added a try/catch around the `consume(msg)`. It's just strange that its exactly after 1 minute, which is exactly the `pongWait` value of bettercap. 
https://github.com/bettercap/bettercap/blob/4eead7eafaec57282ad435473e420d7bad7a602e/modules/api_rest/api_rest_ws.go#L18

Websockets is handling the ping/pong stuff internaly. There was an issue in 7.x versions, which is why I upgraded to 8.1. Yesterday I was just playing with the websocket.connect options, like `ping_interval` and `ping_timeout`, but I could not fix it! But today is another day!!

dadav, 2020-04-08:
```shell
DEBUG:websockets.protocol:client < Frame(fin=True, opcode=9, data=b'', rsv1=False, rsv2=False, rsv3=False)
[2020-04-08 08:14:25,107] [DEBUG] client < Frame(fin=True, opcode=9, data=b'', rsv1=False, rsv2=False, rsv3=False)
DEBUG:websockets.protocol:client - received ping, sending pong: [empty]
[2020-04-08 08:14:25,124] [DEBUG] client - received ping, sending pong: [empty]                                                                  
DEBUG:websockets.protocol:client > Frame(fin=True, opcode=10, data=b'', rsv1=False, rsv2=False, rsv3=False)
[2020-04-08 08:14:25,179] [DEBUG] client > Frame(fin=True, opcode=10, data=b'', rsv1=False, rsv2=False, rsv3=False)
# some more data
# and then...     
DEBUG:websockets.protocol:client - event = eof_received()                                                                                        [2020-04-08 08:14:25,216] [DEBUG] client - event = eof_received()                                                                                DEBUG:websockets.protocol:client ! failing OPEN WebSocket connection with code 1006                                                              
[2020-04-08 08:14:25,321] [DEBUG] client ! failing OPEN WebSocket connection with code 1006                                                      
DEBUG:websockets.protocol:client - event = connection_lost(None)
[2020-04-08 08:14:25,343] [DEBUG] client - event = connection_lost(None) 
DEBUG:websockets.protocol:client - state = CLOSED
[2020-04-08 08:14:25,410] [DEBUG] client - state = CLOSED
DEBUG:websockets.protocol:client x code = 1006, reason = [no reason]
```

evilsocket, 2020-04-08:
you might want to print the error here https://github.com/bettercap/bettercap/blob/4eead7eafaec57282ad435473e420d7bad7a602e/modules/api_rest/api_rest_ws.go#L98

evilsocket, 2020-04-08:
pushed a more verbose log of the error, hope that helps

dadav, 2020-04-08:
> 
> 
> pushed a more verbose log of the error, hope that helps

thx!!!

seems to fail because of i/o timeout >.>

dadav, 2020-04-08:
ok, i think i got it! Will push a fix durring the day. Seems like the await consumer(msg) blocks everything, even the pong-functionality.

dadav, 2020-04-08:
Shouldnt here:
https://github.com/bettercap/bettercap/blob/4eead7eafaec57282ad435473e420d7bad7a602e/modules/api_rest/api_rest_ws.go#L42

also do:
`ws.SetReadDeadline(time.Now().Add(pongWait))` 
?

Otherwise e.g. for the initial ping (after 54secs), pwnagotchi only has 6 secs (ReadTimeout was set to 60secs and 54secs are already over) to answer it with a pong. This is normaly easily doable, but at the beginning there are a sh!tload of events which keep pwnagotchi busy and therefore it takes longer than 6secs to send the first pong.

evilsocket, 2020-04-08:
possibly

PR #845 (2020-04-13): ready to merge

- fixed imports (especially if you just want to get the version or work with plugins)
- fixed chrome issue (fixed #819)
- fixed the output of some error msgs
- fixed config output (sections are now separated by newline)
- fixed thread problem (fixed #793)
- added sad/bored states to reward function (fixed #51)
- added on_config_changed cb (fixed #822)
- added reward curve to session-stats plugin
- added logtail plugin
- added event polling via websocket (fixed #43)
- added plugins subcommand (fixed #105)
- added remove_whitelisted function (closes #734)
- added whitelist functi [... cut]
closed 2020-04-13 by evilsocket