evilsocket/pwnagotchi #45: implement optional BT connection and possibly tethering

evilsocket, 2019-10-02:
implement optional BT connection and possibly tethering in order to reach the UI from a smartphone or tablet without wires

evilsocket, 2019-10-02:
@caquino cc

caquino, 2019-10-02:
@evilsocket I managed to get Bluetooth and tethering working but hit a wall regarding MDNs, according to all the information I found MDNS is not supported over Bluetooth PAN because it uses a different encapsulation that is not really TCP/IP.

IOs devices have custom workarounds on top of PAN that allows MDNS/Bonjour/Zeroconf to work, but this is exclusive to IOs devices.

With that being said, the issue is that the Bluetooth host runs a DHCP server so the IP address of the pwnagotchi (Bluetooth Client) is unknown, which would make it hard for us to be able to connect to it.

For this to be usable, it would need us or to show the device IP address on the device screen for the user type on the browser, or to code a mobile app that does some sort of device discovery.

Another option is to use an OTG cable and connect the USB gadget mode to the phone USB (tested and working on IOs and Android), but it's not wireless which is not the same solution.

evilsocket, 2019-10-02:
can we use a static address instead? but yeah showing the ip on the screen is an option when BT connectivity is detected ... idk

caquino, 2019-10-02:
Well, at the moment I can't see a way for us to do that, it would not work because each device uses different network blocks for the PAN network, so we would need to create a static IP from the same network for it to work, which would require previous knowledge from the device network.

Maybe we could negotiate DHCP to get the information and them configure a static IP of that same network, but the issue is that you can't see on most devices which IP address is used on that network, so the user would still not know which IP to connect to.

I tried to find the network information on my iPad OS and on my Android, and they were not available anywhere that is easily accessible by the user.

benleb, 2019-10-02:
In my setup i use a static ip on a bridge with usb0 and bnep0 connected to it. maybe this would also be a nice option here?

(btw. this solution emerged from the PITA project ;) this way its a little more flexibel and bettercap can utilize the bt interface.)

evilsocket, 2019-10-02:
i'm so confused by all the RPI configurations i've been using ...

caquino, 2019-10-03:
Derek assign: caquino

evilsocket, 2019-10-15:
well, i'd say that @dadav nailed this :D

closed 2019-10-15 by evilsocket

LaurentFough, 2019-11-07:
* Not trying to re-open; but searching for a solid solution for BT-PAN/ BT-Tethering on macOS.

> In my setup i use a static ip on a bridge with usb0 and bnep0 connected to it. maybe this would also be a nice option here?
> 
> (btw. this solution emerged from the PITA project ;) this way its a little more flexibel and bettercap can utilize the bt interface.)

@benleb Can you provide more information on this setup?