evilsocket/pwnagotchi #47: use BMP files for the faces

evilsocket, 2019-10-02:
By using a bmp instead of just text for faces, we can render whatever we want, while now we have limitations due to what the font and generally ascii emoji support

dipsylala, 2019-10-02:
This could also more easily open us up to Pwnagotchi themes, or different Pwnagtochi 'creatures'

JRWR, 2019-10-04:
Randomized "creatures" based off whats around it

hmax42, 2019-10-05:
please keep the text based faces as an option.
they have a special charm.

evilsocket, 2019-10-05:
ahaha maybe that can be an option! the idea is to have "themes" so anyone can customize their unit easily :D

lovelaced, 2019-10-26:
+1 for this (and +1 for the default being the text-based faces); it would be really nice for us with tricolor displays to be able to use custom images to utilize color.

evilsocket, 2019-10-26:
it's not that difficult to do

roodriiigooo, 2023-05-03:
Plz finish this feature <3