evilsocket/pwnagotchi #55: simple plugin engine/hook for users on capturing a handshake

xenDE, 2019-10-02:
Develop a simple plugin engine for users to add their own functions after capturing a handshake.

for example:

 - get a GPS position from a GPS module and save the position to root/handshakes/ap.pcap.gps
 - save names of nearly APs
 - ...

example handshake:
192.168.2.0/24 > 192.168.2.2  » [13:57:31] [wifi.client.handshake] captured 84:d6:d0:b4:67:74 -> EasyBox-474250 (e0:60:66:23:61:79) WPA2 handshake (half) to /root/handshakes/EasyBox474250_e06066236179.pcap

create file:
/root/handshakes/EasyBox474250_e06066236179.pcap.gps

evilsocket, 2019-10-02:
this doesn't need to be a secondary engine, it's a duplicate of #54

closed 2019-10-02 by evilsocket

evilsocket, 2019-10-02:
closed together with #54