evilsocket/pwnagotchi #56: -S for the bmap-tool is not completely done

benleb, 2019-10-02:
`./create_sibling.sh: illegal option -- S`

seems like the option is documented but not (fully) implemented. probably this was missed in https://github.com/evilsocket/pwnagotchi/pull/41 ?

caquino, 2019-10-02:
cc @dadav is this something that we need to keep open for you to track?

massar, 2019-10-02:
no need to use -S for sparse, as it detects the mode (well, it detects if bmaptool is available).

benleb, 2019-10-02:
yes and this works like expected 👍 but if there is a -S option in the doc, the tool should recognize it 😅

massar, 2019-10-02:
I'll submit a patch to remove it ;)

caquino, 2019-10-03:
We are moving to a new builder where create_sibling will not be needed anymore, and because of that I'm closing this ticket.

Thanks for reporting the issue!

caquino, 2019-10-03:
Derek close

closed 2019-10-03 by derek[bot]