evilsocket/pwnagotchi #59: consider using a localization platform

alwayslivid, 2019-10-02:
The use of a localization platform (such as Transifex or Crowdin) is a better way of having multiple people work on translations in all sorts of languages together, hence the reason why I'm proposing the integration of this repository in either one of those platforms.

A few of the benefits would be:
- better organization.
- less PRs to deal with/automation features.
- easier for people (and outsiders that haven't looked into the codebase) to contribute to/does not require knowledge of Git, which could result in the attraction of a wider base of volunteers.

caquino, 2019-10-03:
Derek add label: idea

evilsocket, 2019-10-05:
cc @dadav to keep track of the conversation