evilsocket/pwnagotchi #110: Add a plugin to upload APs to wigle.net

dadav, 2019-10-03:
Idea from @evilsocket  ^^

dadav, 2019-10-03:
Derek add label: todo

akosednar, 2019-10-05:
I'll start tackling this.

akosednar, 2019-10-05:
Derek assign: akosednar

akosednar, 2019-10-05:
Derek add assign: akosednar

evilsocket, 2019-10-05:
you are not in the .DEREK.yml list, it won't work :D

akosednar, 2019-10-05:
Ah makes sense :P

Bedrovelsen, 2019-10-06:
Since they have an API these days it is quite quick to implement 

https://api.wigle.net/swagger#/Network_observation_file_upload_and_status./upload

```
$> curl -X POST "https://api.wigle.net/api/v2/file/upload" -H "accept: application/json" -H "Content-Type: multipart/form-data" -F "file=@pwnagotchi.log;type=" -F "donate=true”
```
returns 
```
{
  "success": true,
  "warning": null,
  "results": {
    "timeTaken": "0",
    "filesize": 345025,
    "filename": "1570354259_pwnagotchi.log",
    "transids": [
      {
        "file": "1570354259_pwnagotchi.log",
        "size": 345025,
        "transId": "20191006-00094"
      }
    ]
  },
  "observer": "anonymous"
}
```

Headers
```
 access-control-allow-credentials: true 
 access-control-allow-origin: https://api.wigle.net 
 connection: keep-alive 
 content-length: 230 
 content-type: application/json 
 date: Sun, 06 Oct 2019 09:30:59 GMT 
 server: nginx/1.14.1 
 strict-transport-security: max-age=63072000; includeSubdomains; preload 
```

closed 2019-10-06 by dadav