evilsocket/pwnagotchi #112: [BUG] create_sibling crash when using single cpu vm.

tisboyo, 2019-10-03:
**Describe the bug**
Running Ubuntu 18.10 in a VirtualBox VM with a single cpu causes an error in create_sibling.sh on line 208. Changing the line to taskset -c 0 from taskset -c 1 allows the script to continue running. 

**To Reproduce**
Steps to reproduce the behavior:
1. ... Run Ubuntu in VM with single CPU
2. ... Run create_sibling.sh

**Expected behavior**
Expect script to finish build.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment (please complete the following information):**
 - Pwnagotchi version - commit 58edfd2
 - OS version - Ubuntu 18.10 in VM
 - Type of hardware 
 - Any additional hardware used

**Additional context**
I did not test changing the VM to 2 CPU's, but I imagine that would also be a viable fix.

dadav, 2019-10-06:
Thanks for your report! We are currently working on a replacement and won't put any further effort into maintaining this script. Please use the - soon to be availaible - installation method ^.^

dadav, 2019-10-06:
Derek close

closed 2019-10-06 by derek[bot]