evilsocket/pwnagotchi #124: [Guide]Use a Wavehare 3.5In SPI TFT Screen

JRWR, 2019-10-03:
- First, SSH into your pwnagotchi, and give it some internet! Don't forget to check your default gateway and apt-get update
- Follow Guide here: https://www.waveshare.com/wiki/3.5inch_RPi_LCD_(A)#Method_1._Driver_installation
- At the step with ./LCD35-show add lite to the command prompt (./LCD35-show lite)
- Reboot
- as root, make three symlinks, cd ~; ln -s pwnagotchi.png pwnagotchi_1.png; ln -s pwnagotchi.png pwnagotchi_2.png; ln -s pwnagotchi.png pwnagotchi_3.png;
- apt install fbi
- change display type to "inky" in config.yml
- edit /etc/rc.local and add
`fbi -T 1 -a -noverbose -t 15 -cachemem 0 /root/pwnagotchi_1.png /root/pwnagotchi_2.png /root/pwnagotchi_3.png &`

reboot and you should be good, some of this guide will work with other framebuffer based displays!

evilsocket, 2019-10-05:
@hexwaxwing mind adding this to the docs?

hexwaxwing, 2019-10-05:
Derek assign: hexwaxwing

evilsocket, 2019-10-05:
closed by @hexwaxwing

closed 2019-10-05 by evilsocket

michelep, 2019-10-18:
Following this guide, it show briefly when booted-up and then, after few seconds, show the login prompt. Something wrong?

michelep, 2019-10-18:
Following this guide LCD displays phwagotchi just for few seconds before login prompt: i've solved using the autologin features (https://www.opentechguides.com/how-to/article/raspberry-pi/5/raspberry-pi-auto-start.html) and changing the line to add to /etc/rc.local as follow:

ln -s /root/pwnagotchi.png /tmp/pwnagotchi_1.png; ln -s /root/pwnagotchi.png /tmp/pwnagotchi_2.png; ln -s /root/pwnagotchi.png /tmp/pwnagotchi_3.png;

on /etc/profile i've added, at the end:

. /home/pi/pwnagotchi.sh

and /home/pi/pwnagotchi.sh contains:

#!/bin/bash
fbi -T 1 -a -noverbose -t 15 -cachemem 0 /tmp/pwnagotchi_1.png /tmp/pwnagotchi_2.png /tmp/pwnagotchi_3.png &

Just remember to chmod +x pwnagotchi.sh !

moty66, 2019-10-28:
This hack is really working bad and not clear at all

I suggest to execute this code from  in rc.local  or from /etc/profile  or where ever you want:
```
while true; do 
	fbi -d /dev/fb1 -T 1 -noverbose -a /root/pwnagotchi.png 
	sleep 5
done
```

I didn't load any kernel modules, I didn't edit the /boot/config.txt file, I only installed the driver as described in https://www.waveshare.com/wiki/3.5inch_RPi_LCD_(A)#Method_1._Driver_installation!

michelep, 2019-10-31:
I've worked to a more stable solution, adding support for Waveshare 2,4" LCD display (could be adapted to 3,5") directly in pwnagotchi code. Take a look here: https://www.zerozone.it/tecnologia-e-sicurezza/fatgotchi-pwnagotchi-on-a-24inch-spotpear-display/16589

evilsocket, 2019-10-31:
the "more stable solution" does exactly what the on_frame configuration already does @michelep ... you literally call the fbi binary exactly as the on_frame does

b0n3zx, 2019-11-04:
have any of you got a better refresh rate working on the TFT 3.5 screens yet?

MoJo2600, 2019-11-04:
@b0n3zx you can change the `-t 15` parameter. Currently it is refreshing every 15 seconds. Mine is set to `-t 1` an thus refreshing every second.

michelep, 2019-11-12:
I have definitively solved any refresh issue with a pure python framebuffer memory mapping implementation, that draw directly to fb device without needs of fbi: i got a fps up to 15 without any problem!

agent932, 2019-11-22:
@michelep can you write up a guide to get this setup (I have a generic 3.5 & and a Pi 3) have it working with the above guide but the screen refresh crashes after a while.

234v244, 2019-11-23:
My Raspberry pi 3 cannot access the internet. Can someone help me?

michelep, 2019-11-25:
> @michelep can you write up a guide to get this setup (I have a generic 3.5 & and a Pi 3) have it working with the above guide but the screen refresh crashes after a while.

the patch was included in 1.3.0 release and you did'nt need fbi anymore.

michelep, 2019-11-25:
> My Raspberry pi 3 cannot access the internet. Can someone help me?

check iptables/masq rules (http://www.tldp.org/HOWTO/IP-Masquerade-HOWTO/)

234v244, 2019-11-27:
> Following this guide LCD displays phwagotchi just for few seconds before login prompt: i've solved using the autologin features (use raspi-config) and changing the line to add to /etc/rc.local as follow:
> 
> ln -s /root/pwnagotchi.png /tmp/pwnagotchi_1.png; ln -s /root/pwnagotchi.png /tmp/pwnagotchi_2.png; ln -s /root/pwnagotchi.png /tmp/pwnagotchi_3.png;
> 
> on /etc/profile i've added, at the end:
> 
> . /home/pi/pwnagotchi.sh
> 
> and /home/pi/pwnagotchi.sh contains:
> 
> #!/bin/bash
> sudo fbi -T 1 -a -noverbose -t 15 -cachemem 0 /tmp/pwnagotchi_1.png /tmp/pwnagotchi_2.png /tmp/pwnagotchi_3.png &
> 
> Just remember to chmod +x pwnagotchi.sh !

I did exactly the instructions. But my LCD screen does not refresh over time.

elhenro, 2019-12-10:
@michelep thank you! i can confirm that only using your **more stable** (and cleaner imo) solution, I got it running without issues with the waveshare 3.5 display, with pi zero w and also with pi 3b+.

But somehow the step with `sudo raspi-config` -> `autologin` broke my usb connection mode, the network interface does not come up anymore. Any ideas why this happened maybe?

agent932, 2020-04-15:
> > @michelep can you write up a guide to get this setup (I have a generic 3.5 & and a Pi 3) have it working with the above guide but the screen refresh crashes after a while.
> 
> the patch was included in 1.3.0 release and you didn't need fbi anymore.

Is there a way to add the waveshare 3.5 config as a display option in the config without going through all of this, i am not understanding "the patch was included in 1.3.0 release and you didn't need fbi anymore."

thefunkygibbon, 2020-08-14:
does this still need to be done given that i already seem to have FBI installed and also the config.txt/cmdline.txt changes already exist?

agent932, 2020-12-02:
I spent the day going through all the 3.5 LCD info I could find and none of these are currently working. Is there an updated method?

waldeckerthefirst, 2021-09-24:
hello when i try the guide i get now that the file or directory does not exist for pwnagotchi.png and i could not find it on the FS

condor13, 2023-11-04:
> i added support for this display, see pull request #1014

Hello, 
Sorry kinda noob here... how do I implement this display? All works..web UI works, display works but it doesnt display the UI.. just console. I made it autologin...no luck. Any help? Thanks

h4km4k, 2023-11-04:
Hello, you have to install the waveshare display driver too.

`git clone https://github.com/waveshare/LCD-show.git`

`cd LCD-show/`

`chmod +x LCD35-show`

`./LCD35-show`

And of course your pwnag. needs to be online for that.

bicurico, 2023-12-26:
It took me 2 days to get this to work.
I am using a Raspberry Pi Zero W with a Waveshare 3.5" screen (A) v3.
Ghost's list is almost perfect, but there were a few hickups:

1) The pwnagotchi.png is not stored to \tmp but into \var\tmp. These must be corrected:

ln -s /root/pwnagotchi.png /var/tmp/pwnagotchi_1.png; ln -s /root/pwnagotchi.png /var/tmp/pwnagotchi_2.png; ln -s /var/root/pwnagotchi.png /tmp/pwnagotchi_3.png;

2) You need to set the AUTOLOGIN: sudo raspi-config -> autologin 

3) Where it says:

nano /etc/rc.local

Add ln -s /root/pwnagotchi.png /tmp/pwnagotchi_1.png; ln -s /root/pwnagotchi.png /tmp/pwnagotchi_2.png; ln -s /root/pwnagotchi.png /tmp/pwnagotchi_3.png;
then save.

This must be put BEFORE the exit 0 line, otherwise the links are never created, because the script exists at the exit 0...

I think that was it.

Cheers

wink05, 2023-12-27:
> It took me 2 days to get this to work. I am using a Raspberry Pi Zero W with a Waveshare 3.5" screen (A) v3. Ghost's list is almost perfect, but there were a few hickups:
>
> 1. The pwnagotchi.png is [...]

how you managed to fix issue when you install LCD35-show driver you no longer can ssh into pwnagotchi?

bicurico, 2023-12-27:
> Não costuma receber e-mails de ***@***.*** Saiba por que motivo isto é importante<https://aka.ms/LearnAboutSenderIdentification>

Note: my previous instructions may not be 100% correct. I will post the correct symbolic links later on.
Also, I wrote an improved pwnagotchi.sh that restarts FBI automatically when it crashes.

If the ssh is not working, you need to edit the two files in /boot as instructed. If you edited them, note that running the LCD command will undo your edit.

Enviado de Outlook para Android<https://aka.ms/AAb9ysg>

  1.  You need to set the AUTOLOGIN: sudo raspi-config -> autologin
  2.  Where it says:

bicurico, 2023-12-29:
This is my pwnagoshi.sh script:

`
#!/bin/bash
START=true

while $START
do
  readarray -t PIDS < <(exec pgrep -x fbi)
  if [[ ${#PIDS[@]} -eq 0 ]]; then
    echo "No fbi process was found."
    sudo fbi -T 1 -a -noverbose -t 5 -cachemem 0 /tmp/pwnagotchi_1.png /tmp/pwnagotchi_2.png /tmp/pwnagotchi_3.png
  elif ps -fp "${PIDS[@]}" | fgrep -F '<defunct>' >/dev/null; then
    echo "Defunct fbi process found: ${PIDS[*]}"
    sudo kill -s SIGTERM "${PIDS[@]}"
    sudo kill -s SIGKILL "${PIDS[@]}"
  fi
  sleep 15
done
`
My rc.local looks like this:

`
#!/bin/sh -e
#
# rc.local
#
# This script is executed at the end of each multiuser runlevel.
# Make sure that the script will "exit 0" on success or any other
# value on error.
#
# In order to enable or disable this script just change the execution
# bits.
#
# By default this script does nothing.

# Print the IP address
_IP=$(hostname -I) || true
if [ "$_IP" ]; then
  printf "My IP address is %s\n" "$_IP"
fi

ln -s /var/tmp/pwnagotchi/pwnagotchi.png /tmp/pwnagotchi_1.png
ln -s /var/tmp/pwnagotchi/pwnagotchi.png /tmp/pwnagotchi_2.png
ln -s /var/tmp/pwnagotchi/pwnagotchi.png /tmp/pwnagotchi_3.png

exit 0
`

In order to edit stuff as root, it is necessary to do `sudo su`.

The rest is by the book, as described by Ghost.

condor13, 2024-01-11:
> This is my pwnagoshi.sh script:
> 
> ```shell
> #!/bin/bash
> START=true
> 
> while $START
> do
>   readarray -t PIDS < <(exec pgrep -x fbi)
>   if [[ ${#PIDS[@]} -eq 0 ]]; then
>     echo "No fbi process was found."
>     sudo fbi -T 1 -a -noverbose -t 5 -cachemem 0 /tmp/pwnagotchi_1.png /tmp/pwnagotchi_2.png /tmp/pwnagotchi_3.png
>   elif ps -fp "${PIDS[@]}" | fgrep -F '<defunct>' >/dev/null; then
>     echo "Defunct fbi process found: ${PIDS[*]}"
>     sudo kill -s SIGTERM "${PIDS[@]}"
>     sudo kill -s SIGKILL "${PIDS[@]}"
>   fi
>   sleep 15
> done
> ```
> 
> My rc.local looks like this:
> 
> ```shell
> #!/bin/sh -e
> #
> # rc.local
> #
> # This script is executed at the end of each multiuser runlevel.
> # Make sure that the script will "exit 0" on success or any other
> # value on error.
> #
> # In order to enable or disable this script just change the execution
> # bits.
> #
> # By default this script does nothing.
> 
> # Print the IP address
> _IP=$(hostname -I) || true
> if [ "$_IP" ]; then
>   printf "My IP address is %s\n" "$_IP"
> fi
> 
> ln -s /var/tmp/pwnagotchi/pwnagotchi.png /tmp/pwnagotchi_1.png
> ln -s /var/tmp/pwnagotchi/pwnagotchi.png /tmp/pwnagotchi_2.png
> ln -s /var/tmp/pwnagotchi/pwnagotchi.png /tmp/pwnagotchi_3.png
> 
> exit 0
> ```
> 
> In order to edit stuff as root, it is necessary to do `sudo su`.
> 
> The rest is by the book, as described by Ghost.

Thanks for this. I got it working on a waveshare 3.5inch. 
I am testing this with a PI Zero (no wifi no BT) until PI zero W 2 arrives. 
This little one doesnt seem to get past the "Generating Keys" :)
So its not a good idea to try this old beast :)

phaake37, 2024-07-26:
THANK YOU @condor13 - I spent hours trying to get this working and your scripts were the key.

condor13, 2024-12-20:
> THANK YOU @condor13 - I spent hours trying to get this working and your scripts were the key.

Hihi.. learned from the best :)