evilsocket/pwnagotchi #137: [BUG] Round / truncate times in UI

deveth0, 2019-10-04:
**Describe the bug**
When the pwnagotchi is sleeping, it displays the remaining time. The time is not rounded or truncated, so it might be something like `(13.9999999999s)` which is too long to display. 

**To Reproduce**
Steps to reproduce the behavior:
1. Run a pwnagotchi
2. Wait for it to sleep
3. Check the time

**Expected behavior**
I'd expect that the remaining time is displayed as a integer and not a float.

**Environment (please complete the following information):**
 - Pwnagotchi version: build 5968030ed
 - OS version: create_sibling.sh created raspbian
 - Type of hardware: Pi0W with waveshare v1

evilsocket, 2019-10-04:
fixed already

closed 2019-10-04 by evilsocket

deveth0, 2019-10-04:
kwl, thx