evilsocket/pwnagotchi #138: pwnchinko! esp based pwnagotchi companion

evilsocket, 2019-10-04:
We should do something very simple but effective. An ESP should listen for management frames and reinject them as they are, this will effectively boost their rssi levels and, if the pwnagotchi is in range, it will detect them from the ESP itself, which would add as kind of a bridge.

The ESP should keep a list of the APs being "proxied".

Whenever pwnagotchi will deauthenticate or associate to one of those stations, the ESP will detect this packets as well, "remember" that they contain elements that they previously proxied and perform the same operation, reinject the deauths and association frames, along with the resulting (if any) handshake packets.

In a second iteration of this, given that pwnagotchi's identity is RSA based, we might want to experiment and see if ESP can do RSA encryption at the firmware level, so that only the target pwnagotchi would be able to decrypt the proxied frames.

dadav, 2019-10-04:
why whould you choose "chinko"? it means penis in japanese LOL

evilsocket, 2019-10-04:
it's @caquino's idea! :D but i think it makes sense, it's like a small(er) wifi cock for pwnagotchi XD

caquino, 2019-10-05:
the idea actually was because of [pachinko](https://en.wikipedia.org/wiki/Pachinko)

hmax42, 2019-10-13:
do the esp and the pwnagotchi communicate?
since both can listen only on one channel, channel changes should happen synchonous, right?

MischaBoender, 2019-10-16:
> do the esp and the pwnagotchi communicate?
> since both can listen only on one channel, channel changes should happen synchonous, right?

They could connect / sync to each other over bluetooth.

hmax42, 2019-10-18:
would the wifi channel not be an issue?