evilsocket/pwnagotchi #141: [BUG] Python OSError 105 (ENOBUFS)

grufwub, 2019-10-05:
```[2019-10-05 00:22:02,845] [ERROR] error
Traceback (most recent call last):
  File "/root/pwnagotchi/scripts/pwnagotchi/mesh/advertise.py", line 89, in _sender
    sendp(self._frame, iface=self._iface, verbose=False, count=5, inter=self._period)
  File "/usr/local/lib/python3.7/dist-packages/scapy/sendrecv.py", line 338, in sendp
    realtime=realtime, return_packets=return_packets)
  File "/usr/local/lib/python3.7/dist-packages/scapy/sendrecv.py", line 291, in __gen_send
    s.send(p)
  File "/usr/local/lib/python3.7/dist-packages/scapy/arch/linux.py", line 559, in send
    return SuperSocket.send(self, x)
  File "/usr/local/lib/python3.7/dist-packages/scapy/supersocket.py", line 49, in send
    sent = self.outs.send(sx)
OSError: [Errno 105] No buffer space available
```

After about an hour of running I started getting this error repeatedly, only fixed with a reboot. Left her running over night then came back to check logs in the morning and found this 😛 

I'm not sure if this more a unique / device specific bug (I'm running on an RPi3b+), but there's some suggestions to limit number of concurrent threads in this thread where someone was having the same issue: https://stackoverflow.com/questions/31746401/python-loop-causing-error-errno-105-no-buffer-space-available

evilsocket, 2019-10-05:
i never had this issue myself so i can't test, but i pushed a possible fix, @grufwub mind testing to see if it happens again after you updated the codebase? thanks! :D

closed 2019-10-05 by evilsocket

grufwub, 2019-10-05:
been testing this out for a walk this evening -- doesn't seem to be running into this issue anymore, though there were a lot of messages in BetterCap itself about being unable to inject packet due to resource WiFi unavailable 🤔

also regarding the fix -- is this error being produced just worth ignoring then??

evilsocket, 2019-10-05:
it's good info to have in case you're reading the log to understand what's going on, but not a critical error, it only means the wifi chipset is stressed

commit 03094264 (2019-10-05): fix: possible fix for ENOBUFS error while advertising (fixes #141)