evilsocket/pwnagotchi #155: x86 32-bit Support

grufwub, 2019-10-05:
Spent today trying to get Pwnagotchi working on a certain 32-bit x86 tablet-esque device... Ran into some issues with libjasper-dev not existing in Debian repositories for 32-bit Linux, same with tensorflow in pip repositories.

Getting this working is *probably* possible with some one-off compiles of Jasper and Tensorflow for 32bit (okay I'm not 100% sure how possible this is but I'd just like to assume 😛). So while support 32-bit x86 officially may not be worth it, it could be worth just hosting some builds of the required libraries. That way if people do want to put in the leg-work to get it working, then they don't have to spend ages waiting on library compiles. What do you think?

Though I'm yet to get it working, and will have to spend more time on it tomorrow.

evilsocket, 2019-10-05:
hosting the libraries (especially TF given that it includes several native things) would be cool, but who's gonna build them? :D

grufwub, 2019-10-05:
i'll give it a go tomorrow see how it goes :3

evilsocket, 2019-10-05:
good luck! :D

gh0stshell, 2019-10-09:
Found an i386 repo that has libjasper, maybe this will work for you

Repo :Trinity R14 BuildDeps i386

deb http://ftp.fau.de/trinity/trinity-builddeps-r14.0.0/debian/ stretch main
sudo apt-key adv --keyserver keyserver.quickbuild.io --recv-keys F5CFC95C
sudo apt-get update
sudo apt-get install libjasper-dev

l0ophole, 2019-10-26:
I got it installed on x86/64 but I get this error when I run the binary:

```

RuntimeError: This module can only be run on a Raspberry Pi!

```

lazee486, 2019-11-04:
go to the config file and turn video true to false