evilsocket/pwnagotchi #160: find and document a way to have a RTC working via i2c

evilsocket, 2019-10-05:
find and document a way to have a RTC working via I2C

caquino, 2019-10-08:
Derek add label: Hacktoberfest

leon-th, 2019-10-09:
Should the RTC work as plugin? Or how should it work?

leon-th, 2019-10-10:
I programmed it like this:
I have a script that enables or disables the RTC settings. If you have a clock and disconnect without 
configuring it, the system returns to 1970. To prevent this, the script check called by a plugin is done 
at startup to see if the clock is connected.
The plugin starts the script at startup and synchronizes the time when the internet is available
Is it good like that?

LeonT

hmax42, 2019-10-13:
since there is an optional gps plugin, would it be feasible to acquire the time from the gps?