evilsocket/pwnagotchi #166: find a way to optimize AI loading times

evilsocket, 2019-10-05:
TensorFlow takes minutes to import on a Raspberry Pi Zero W and that's probably because of the huge .so file with native primitives it has to load, among other things. Given the nature of the project, that stuff is imported only once, so caching it in memory wouldn't speed things up. Switching frameworks is not feasible, unless we have the same exact features (unlikely given that stable-baselines is TF based). For instance, there's no stable-baselines port for TF-lite.

evilsocket, 2019-10-05:
@caquino this is a nice hacktoberfest entry

evilsocket, 2019-10-15:
@araffin thanks for your feedback!

* Yep TF alone takes a lot, mostly due to a 130MB .so file with all the native code.
* The LSTM seemed to perform way better than MLP alone, it tends to generate useful policies way faster, that is why the choice. As long as I know, if I don't use VecFrameStack an exception is thrown. Will check PPO2, thanks a lot!

For #49 I already have my own implementation, but thanks :D

d3sm0, 2019-10-28:
Importing stable baseline, all old tensorflow (works on v1 not v2) which loads all the tf graph in memory the first time, instead of compiling at run time, makes way to heavy. 

Maybe one can rewrite a2c  tuned for this application and use  (not sure if you are already doing this) a compiled version for rasberry. This should reduce loading time quite a bit.

As a side: LSTM is computationally heavy to run (complexity is on the size of the features)  and it make sense if you have long time dependencies in your time-series. Is this the case? One can try to use 1d convolution (complexity is on number of samples)  which are more suitable for time-series kind of signal.

evilsocket, 2019-10-28:
> As a side: LSTM is computationally heavy to run (complexity is on the size of the features) and it make sense if you have long time dependencies in your time-series. Is this the case?

yes