evilsocket/pwnagotchi #169: Inkyphat/inky display

grymoire, 2019-10-05:
Support for the Inkyphat display is missing, despite the documentation stating it's supported.

Bedrovelsen, 2019-10-06:
Have you set the correct display option in the [ui.display](https://github.com/evilsocket/pwnagotchi/blob/8b4350045e3cd298281dffba903f0b4fea90b76e/pwnagotchi/defaults.yml#L130) area of the config file?

```
Set the type of display you want to use via ui.display.type (if your display does not work after changing this setting, you might need to completely remove power from the Raspberry and make a clean boot).

You can configure the refresh interval of the display via ui.fps. We recommend using a slow refresh rate to avoid shortening the lifetime of your e-ink display. The default value is 0, which will only refresh when changes are made to the screen.
```

Can you elaborate on what you mean by the support is  missing?

I should be soldering  header pins to my pi0w’s GPIO tomorrow at some point as I have an Inky I will be using for the project.  If it is missing as you have submitted then I should get a similar outcome. If I have no issues when setting it up on my end however I will gladly help you fix the issue you are having.

swedishmike, 2019-10-06:
I have an Inky on its way - should be with me tomorrow or on Tuesday - and am happy to try and replicate/fix once it's with me as well if the issue is still open.

@grymoire - since the libraries are installed during the image build, does your display work using the examples on Pimoronis site/Github repo? Just to make sure it's not an issue with the display itself.

nachogarciaegea, 2019-10-06:
It works for me with examples of Pimoronis / Github repository, but I have not been able to make it work with pwnagotchi...

natur3kid, 2019-10-06:
It works flawless. You just have to set it in config.yml.

grymoire, 2019-10-06:
I changed the config

```
# grep -3 inky config.yml
    display:
        enabled: true
        rotation: 180
        # Possible options inkyphat/inky, papirus/papi, waveshare_1/ws_1 or waveshare_2/ws_2
        type: 'inkyphat/inky'
        # Possible options red/yellow/black (black used for monocromatic displays)
        color: 'black'
        video:

```
And my log file reports
`[CRITICAL] unknown display type inkyphat/inky`

Also - I note that unlike the other two displays ui/papirus and ui/waveshare,  there is no directory  for the inky display. I looked for the library based on this line, 
`from inky import InkyPHAT`
but I didn't see any file that matched. 
I'm willing to work on this, but I'm not sure how to restart the display without rebooting the entire box.

swedishmike, 2019-10-06:
@grymoire  I believe that you can use either inkyphat or inky - the slash is there to denote that so to speak.

From view.py:  `if config['ui']['display']['type'] in ('inky', 'inkyphat'):`

grymoire, 2019-10-06:
Ah. That was it! The comment was confusing. 
As an aside, I'm not sure if the custom.yml has just the difference/update over config.yml, or if it replaces config.yml. It would be nice to have a sample custom.yml that has everything commented out, for newbies like me.

closed 2019-10-06 by grymoire