evilsocket/pwnagotchi #170: Missing SDCARD folder, causing create to fail [BUG]

gh0stshell, 2019-10-06:
Went through the history of the repo and on 10/4 SDCARD was in the repo and at some point it went missing, causing the rsync on lines 175/176 to fail causing the create script to fail. Not sure if there is a reason for its removal. To solve it for me I copied it from an archeive repo i stored of the last passing build. I figure its easier for you to push it then for me to do a pull just in case sdcard was removed for a reason.

incredincomp, 2019-10-06:
Lines in script referenced here. https://github.com/evilsocket/pwnagotchi/blob/8b4350045e3cd298281dffba903f0b4fea90b76e/scripts/create_sibling.sh#L175

gh0stshell, 2019-10-06:
Even with the sdcard rsync ran into some issues, uploading a fix now to my repo in case the removal of sdcard was an overlook and included a screen shot of the error that happens, just confirming the fix first and making sure sdcard was in fact suppose to still be in the repo

gh0stshell, 2019-10-06:
It worked and the image built successfully, as a temp fix I created a repo for the script to download with the newest pwnagotchi data with the sdcard folder added to it since having it try to copy from the host was not working to well:

https://github.com/gh0stshell/gotchi.git pwnagotchi

specified the folder since the script looks for pwnagotchi

it then was able to download the needed repo with the sdcard folder and finish for me at least

dadav, 2019-10-06:
Derek close

closed 2019-10-06 by derek[bot]