evilsocket/pwnagotchi #192: what about a "wifi location" plugin?

zenzen666, 2019-10-06:
A plugin that saves the "wifi location" of every handshake captured in a json file with the mac address and signal of the strongest aps. 
The file can then be used with Google Geolocation API or Mozilla Location Service to retrieve the geographic location of the event.

Possible evolution: 
- Save the last location in a wifi location history file, usable to build a path map.
- Send the last location to an external server through dns tunnel when conditions are good (not busy and with an access point with dns open available. Lot of cities have access points with paywall but the dns still usable). Something like "find my pwnagotchi" LOL

I know that location services can be implemented easily using a gps module or a connection to an external device through bluetooth. But a pwnagotchi that can work alone it's more fun!

What people think? If you like the idea I can work on it. The first two steps should be easy.

dadav, 2019-10-06:
I think this was the first plugin ever created for pwnagotchi^^
See https://github.com/evilsocket/pwnagotchi/blob/master/pwnagotchi/plugins/default/gps.py

zenzen666, 2019-10-06:
> See https://github.com/evilsocket/pwnagotchi/blob/master/pwnagotchi/plugins/default/gps.py
yeah but the plugin requires a gps receiver.

the idea is about a way to save location without have a gps receiver or any external device connected to the pwnagotchi.

dadav, 2019-10-06:
Now i understand what you mean! The APs are like fix points and can be used as coordination system. Interesting ^.^

zenzen666, 2019-10-06:
@dadav: yes! It's also they way your phone shows you your location when gps is not available or bad.
Take a loot at:
https://location.services.mozilla.com/api
https://developers.google.com/maps/documentation/geolocation/intro

zenzen666, 2019-10-06:
Here some working code:
https://github.com/evilsocket/pwnagotchi/compare/master...zenzen666:geowifi-plugin
At every new handshake it saves a json file with the mac addresses of the 6 strongest aps.
The file is usable with google or mozilla geolocation api services.

zenzen666, 2019-10-08:
here the version that convert the wifi location in geo location when internet is available
https://github.com/evilsocket/pwnagotchi/pull/211/files

dadav, 2019-10-08:
@zenzen666 Good work^^ I will test it in the evening!!

dadav, 2019-10-08:
We can also use these location data in other plugins like `grid` or `wigle`

dadav, 2019-10-08:
Like, if `.gps.json` is present use it, otherwise check for presence of `.geo.json`

dadav, 2019-10-08:
@evilsocket What do you think?

xenDE, 2019-10-08:
please dont overwrite existing .gps file.
add gps data with timestamp to existing json data.

i see overwriting of the json file in the gps plugin code, instead of adding new gps data.
but i need to check this later, after bring my gps online.

dadav, 2019-10-08:
> please dont overwrite existing .gps file.
> add gps data with timestamp to existing json data.
> 
> i see overwriting of the json file in the gps plugin code, instead of adding new gps data.
> but i need to check this later, after bring my gps online.

they are not overwritten^^

dadav, 2019-10-08:
> Here some working code:
> [master...zenzen666:geowifi-plugin](https://github.com/evilsocket/pwnagotchi/compare/master...zenzen666:geowifi-plugin)
> At every new handshake it saves a json file with the mac addresses of the 6 strongest aps.
> The file is usable with google or mozilla geolocation api services.

check my pr plz

zenzen666, 2019-10-08:
@dadav I tested the code with the last image and everything seems fine. thank you!

zenzen666, 2019-10-09:
PR merged with: https://github.com/evilsocket/pwnagotchi/pull/231
thank you @evilsocket and all the people who helped me

closed 2019-10-09 by zenzen666