evilsocket/pwnagotchi #197: [BUG]

ozzieoffsec, 2019-10-07:
**Describe the bug**

The Update pwnagotchi script is broken due to the git folder structure changing.

The "sdcard" folder has been removed. etc.

The script removes **/root/pwnagotchi** and is unable to restore to the latest GIT version. 

The affected lines I believe are: 

https://github.com/evilsocket/pwnagotchi/blob/6e9bb866c79506b310b3d05374d0d2ec2f87e6d1/scripts/update_pwnagotchi.sh#L101

https://github.com/evilsocket/pwnagotchi/blob/6e9bb866c79506b310b3d05374d0d2ec2f87e6d1/scripts/update_pwnagotchi.sh#L102 

**To Reproduce**

1. Copy script to pwnagotchi
2. Run as per below. 

```root@pwnie:~# ./update_pwnagotchi.sh -h

 usage: ./update_pwnagotchi.sh [OPTIONS]
 Note: This should be run from the pwnagotchi itself!

   Options:
      -v        # Version to update to, can be a branch or commit. (default: master)
      -u        # Url to clone from. (default: https://github.com/evilsocket/pwnagotchi)
      -m        # Mode to restart to. (Supported: auto manual; default: auto)
      -b        # Backup the current pwnagotchi config and hostname references, then overwrite with defaults.
      -r        # Restore the current pwnagotchi config and hostname references after upgrade. (-b will be enabled.)
      -h        # Shows this help.

root@pwnagotchi:~# ./update_pwnagotchi.sh -b -r
[+] Checking prerequisites.
[+] Cloning to /tmp/pwnagotchi...
[+] Creating backup of config.yml and hostname references
[+] Installing 6e9bb86
rsync:** change_dir "/tmp/pwnagotchi/sdcard/boot" failed: No such file or directory (2)
rsync error: some files/attrs were not transferred (see previous errors) (code 23) at main.c(1207) [sender=3.1.3]
rsync: change_dir "/tmp/pwnagotchi/sdcard/rootfs" failed: No such file or directory (2)
rsync error: some files/attrs were not transferred (see previous errors) (code 23) at main.c(1207) [sender=3.1.3]
[+] Restoring backup of config.yml and hostname references
mv: cannot move '/root/config.yml.bak' to '/root/pwnagotchi/config.yml': No such file or directory
[+] Restarting pwnagotchi in auto mode.
[+] Finished
```

**Expected behavior**
The pwnagotchi should update successfully not break. :) 

**Screenshots**

```=rsync:**** change_dir "/tmp/pwnagotchi/sdcard/boot" failed: No such file or directory (2)
rsync error: some files/attrs were not transferred (see previous errors) (code 23) at main.c(1207) [sender=3.1.3]
rsync: change_dir "/tmp/pwnagotchi/sdcard/rootfs" failed: No such file or directory (2)
rsync error: some files/attrs were not transferred (see previous errors) (code 23) at main.c(1207) [sender=3.1.3]
[+] Restoring backup of config.yml and hostname references
mv: cannot move '/root/config.yml.bak' to '/root/pwnagotchi/config.yml': No such file or directory```

**Environment (please complete the following information):**
 - Pwnagotchi version - latest

sidelux, 2019-10-07:
They are refactoring the entire repo, probably you will able to download directly the .img file instead of compile by yourself.

nic0laz, 2019-10-07:
The same is happening on the create_sibling script, no more sdcard directory.

sidelux, 2019-10-08:
Same cause, will be soon available only a prebuild image.

justin-p, 2019-10-08:
Hi @ozzieoffsec, the update_pwnagotchi.sh script was something to make life easier in the earlier stages of the project. The script has been removed from the repo as of commit d86e3cbc272bf69754022a731feaa1184c17cb28.

justin-p, 2019-10-08:
Derek close

closed 2019-10-08 by derek[bot]