evilsocket/pwnagotchi #198: Windows Network Share script not working

HitItAll, 2019-10-07:
Hi, this is most likely my own fault but I can't seem to configure the Windows networking script for pwnagotchi. I've mostly got it to work in ubuntu (the Linux script) but Linux for some odd reason doesn't forward DNS queries (if someone can give me a solution to that, it would be even better). Anyways when I try to use the Windows script in powershell it tells me to configure an IP address but no matter what address I give it says " \win_connection_share.ps1 : X.X.X.X is not a valid IP. (with numbers replacing the Xs of course). Any help would be greatly appreciated.

justin-p, 2019-10-08:
Hi @HitItAll,
 
You shouldn't need to set a specify IP, by default the `-ScopeAddress` switch defaults to `10.0.0.1` when you run `-SetPwnagotchiSubnet`. Also, you only need to run `-SetPwnagotchiSubnet` once. This changes the default subnet Windows uses for internet sharing to be on the pwnagotchi subnet. 

After running the script once with the `-SetPwnagotchiSubnet` switch (and rebooting) you should be able to just run the script with the `-EnableInternetConnectionSharing` switch and everything should be taken care of.

Regarding DNS, look at your resolve.conf. The internet connection sharing scripts don't do anything with DNS, this is all handled by the OS on your pwnagotchi unit.

justin-p, 2019-10-08:
Derek assign: me

brutecode, 2021-03-18:
> You shouldn't need to set a specify IP, by default the `-ScopeAddress` switch defaults to `10.0.0.1` when you run `-SetPwnagotchiSubnet`. Also, you only need to run `-SetPwnagotchiSubnet` once. This [...]

Thanks, that were exactly the directions I needed to set this up. Even after reading the script I couldnt quite wrap my head around this. These steps should be part of the official documentation.