evilsocket/pwnagotchi #201: nearby units ui element should be explained and better rendered

evilsocket, 2019-10-07:
we should explain in the docs that the nearby unit UI element only shows the unit with the strongest signal in that moment even if there're multiple units around. Moreover, some sort of visual indicator should be added (idea from @hexwaxwing) to at least signal the presence of more units (maybe pixels? tiny faces? ...)

dipsylala, 2019-10-07:
Wondering if we could show them this way - it allows the user to see them 'queued' behind each other.

```
(⌐■_■) |||| chiba 11 (13) - just the one friend
((⌐■_■) |||| chiba 11 (13) - two friends
(((⌐■_■) |||| chiba 11 (13) - three friends
```

Pros:
    We don't have to come up with small faces or consider the current status of the other friends. 
    Minimal change to the current display
Cons: 
    May be difficult for some to  see.

evilsocket, 2019-10-07:
and what if there are 100 units? :D it has to be something with a maximum controllable width

dipsylala, 2019-10-07:
We'd hit that with tiny faces/pixels though - if we have that entire line of the display to friends, then we could maybe use this to take us up to 1000?

`
(⌐■_■) |||| chiba 11 (13)  of 999
`

...but we might have to consider truncating hostnames names if the total text length exceeds the screen width

evilsocket, 2019-10-07:
i like that! we can just put "of over 9000" to have a threshold that's also fun! :D

commit 05779728 (2019-10-07): new: if more than one peer are present, display their number (closes #201)
closed 2019-10-07 by evilsocket