evilsocket/pwnagotchi #203: add battery benchmarking to hardware docs

hexwaxwing, 2019-10-07:
we ought to be keeping track of user benchmarked batteries used with a Pwnagotchi, ideally running both with and without AI (to compare).

belongs in **Hardware** section of the docs.

might be organized something like this?:

Manu | Model | mAh | Date tested | Version | Mode | Observed Duration
-------|-------|------|----------|---------|-------|---------------
Anker | PowerCore [A1109](https://www.anker.com/products/variant/powercore-5000/A1109011) | 5000 | 2019-MM-DD | (alpha) | AI | HH:MM:SS
Anker | Astro E7 [A1210](https://www.anker.com/products/variant/astro-e7-26800mah-portable-charger/A1210012) | 25600 | 2019-09-DD | (alpha) | AUTO | 49:MM:SS
PiSugar | [PowerPack L](https://github.com/PiSugar/PiSugar) | 1200 | 2019-10-07 | (alpha) | AI | 04:49:42

hexwaxwing, 2019-10-07:
derek assign: hexwaxwing

hexwaxwing, 2019-10-07:
Derek assign: hexwaxwing

craigsblackie, 2019-10-07:
Manu | Model | mAh | Date tested | Pwnagotchi Version | RPi | Mode | Observed Duration
-- | -- | -- | -- | -- | -- | -- | --
Unbranded | Free from con| 2000 | 2019-10-07 | (alpha) | RPi0W | AI | 08:24:00