evilsocket/pwnagotchi #207: Connect to home WiFi and get DHCP when specific BSSID is detected

sedward5, 2019-10-07:
**Is your feature request related to a problem? Please describe.**
It would be super cool if I didn't need to use USB to SSH to my pwnagotchi. What if, when a predefined BSSID (or BSSIDs) is detected it switches the interface from monitor mode to infrastructure mode, associates with the wifi, and attempts DHCP. If it can't find that network then it could switch the interface back to monitor mode and get to pwning.

**Describe the solution you'd like**
Essentially I would define my home BSSIDs in the the config file, then while scanning, if pwnagotchi discovers one of those network, it brings down the wifi interface and brings it back up in managed mode. Likewise if the device finds itself disconnected from the wifi and that/those BSSID(s) are not available it will bring back down wifi and come back up in monitor mode.

**Describe alternatives you've considered**
My first thought to accomplish this was to add a mechanical switch to change the mode, but this seems easier. I guess bluetooth pairing, or a USB Ethernet adapter could also work.

**Additional context**
The biggest drawback, I think, is that it would be harder to capture near by neighbors wifi handshakes. But I think the easy-of-use gain might outweigh the loss. This is all IMHO of course, feel free to disagree.

cdiemel, 2019-10-09:
What about the use of one of those nano wifi dongles that only gets turned on if the network is noticed?

kpcyrd, 2019-10-09:
> The biggest drawback, I think, is that it would be harder to capture near by neighbors wifi handshakes. But I think the easy-of-use gain might outweigh the loss. This is all IMHO of course, feel free to disagree.

When combined with the backup plugin it could wait for the home ssid to show up, check if a new backup is due, and only in that case switch to infrastructure mode to push a backup and then switch back to pwning afterwards.

This way you could automatically sync captured handshakes somewhere as soon as you get home.

cdiemel, 2019-10-10:
Line 215 of agent.py sorts the APs and discards whitelists, I think that would be a good place to break off and upload the data.  

I guess set a threshold of how often you upload, maybe once an hour.  Then zip all the new stuff, and ftp to a known address on your local network and put the file then maybe remove the local zip? 

With a wlan0 entry in /etc/network/interface.d/wlan0-cgf you can set it to dhcp and to use wpa_supplicant

Real quick script to up the interface: 
```
wpa_supplicant -B -i wlan0 -c /etc/wpa_supplicant/wpa_supplicant.conf
sleep 3
dhclient wlan0
sleep 3
```

Then tearing it down is as easy as 
```ifconfig wlan0 down```

At this point you could return back to processing the APs where you left off.  

I'm not sure if processing the APs is time sensitive and running this section could take a few seconds depending on amount of capture

cd-rum, 2019-10-17:
So in sum to specify a network/networks to not deauth but connect to and upload? +1 for this for sure. Connection-sharing on a RPi0 w/ external wifi is hard enough.

cdiemel, 2019-10-17:
I have gotten it to somewhat work, but the thread in pnagotchi continue running in the background and cause significant errors when the mon interface is down.  I was looking to see if the same could be achieved with a nano adapter

cd-rum, 2019-10-17:
@cdiemel What's your plugin look like right now?

spees, 2019-10-17:
I think this idea could be extended. 

Scenario 1: As you described. Find home wifi and connects. Pretty straight forward.

Scenario 2: The device finds an open network, or a network which it could crack using a dict, and connect to that to upload handshakes and what not. Maybe be able to set a minimum amount of new handshakes it would need before it would switch off monitor mode and actually connect to it. This would also open other possibilities.

cd-rum, 2019-10-17:
@spees In scenario two the RPi0 would have to crack the .pcap itself--my battery ain't that good 😂

spees, 2019-10-17:
If you use a small dict you should be fine. This would only apply to quick wins. You're probably more likely to run into a open network. Especially in crowded areas like shopping malls etc.

Where i live, it is very normal for people to have an additional, separated, open wifi running on their modem.

cdiemel, 2019-10-18:
I haven't pushed it to github yet, I'll try to when I get home.  
Basically I find the matching AP
Take down mon0
Put up wlan0
Wpa-supplicant
Dhclient for a lease and Im connected

But I was filling the logs with capture failures trying to access mon0.  

The one with the external dongle just puts wlan1 up for 2 minutes but I loose Association with my AP after a few seconds and can't get it back without a reboot

evilsocket, 2019-11-12:
this should be a new mode i think ... so that when in AUTO or MANU, if this AP is detected, everything can restart in this CLI mode and the needed services are enabled and started.

cd-rum, 2019-11-12:
Would that expose SSH over wifi? That'd be ideal I think. 

Also it signifies HOME which might mean like it goes into defend mode or something (like alerts when a new AP is detected while not moving)--although it kinda does that already, and I'm not too sure `wlan0` can be in monitor mode while using wifi.

evilsocket, 2019-11-12:
Correct, you'd have the web interface, ssh and everything on your home wifi .... for the defensive mode, i can use the lan modules in bettercap :D

cd-rum, 2019-11-12:
Oh, cool! I had no idea that was possible!

Is that what the defensive mode would do?

sburggsx, 2019-11-12:
I keep thinking of an idea for a version of this. In a "full boat" pwnagotchi, that has a GPS and possibly a better/stronger WiFi adapter, maybe make it possible for the on-board WiFi adapter to be used as a command and control type network. Allow it to attach to different networks (home, phone hotspot) when available but allow the "offensive" WiFi adapter to do it's thing.

SchizoDuckie, 2019-11-14:
It would be great if we can trigger this functionality using a GPIO button when the time has come. That way you could turn it off and on and not have to rely on a detection loop.

cdiemel, 2019-11-28:
@SchizoDuckie it looks like there is availability in the `config.yml` file for GPIO based commands.  

Is `pwnagotchi` callable with arguements?  Something like `pwnagotchi --at-home`.  If that was possible then something like that could be called.  

I also noticed when I ran `touch /root/.pwnagotchi-auto && systemctl restart pwnagotchi` it seemingly restarted my Pi because my SSH session was dropped.  If that is the case, is there a way to do a quick check to see if the specified network is present and drop into "home mode" to SSH and upload, etc.  Then from the SSH CLI you could continue execution and drop the interface into mon mode.  
I think it would make sense for the following start up flow. 

1. check last at home connection (if time is greater than offset time ago drop into loop)
2. if "home" network is present connect
3. provide SSH/FTP/HTTPS/etc
4. when continue command is issued continue execution of pwnagotchi/bettercap
5. go into mon mode and search for APs

From within the loop there is a plugin call to 'on_unfiltered_ap_list' which would allow the ability to see when the specified AP is present and issue the `systemctl restart pwnagotchi` command

troystauffer, 2021-06-02:
I've cobbled together a plugin that accomplishes what was outlined in this issue. It did require a small change in the exception handling of the main loop, which I have opened a PR for. I'll link the plugin I've created as soon as I clean it up a bit.

troystauffer, 2021-06-02:
Still needs some work but this is 90% there: https://github.com/troystauffer/home_base. Props to @c-nagy for most of the commands used here.

michelep, 2021-11-01:
Hi, this is exactly what i'm looking for. Any news about plugin? Can i help in some way?

makerph, 2021-11-03:
the [home based plugin works
](https://github.com/troystauffer/home_base) however found some issue about the epoch cycles still continuing even the mon mode has been disabled.
 
https://github.com/troystauffer/home_base/issues/1