evilsocket/pwnagotchi #229: [BUG] Text overlap on busy screen

dipsylala, 2019-10-08:
The text can sometimes overlap when particular statuses occur:

[image]

Potential changes?
Move the unit name to the top centre of the display, above the top bar. This is a slightly smaller area for the name, but it's a fixed width (assuming max APS detected are 999). This would allow the name to be truncated with '...' if it exceeds 14 characters (based on current font width)

Moving the unit name lets the feedback area grow upwards a line, which should make overlap with the friends bar less likely - it would be able to handle 4 lines without clashing. More space might be gained by making the line spacing less, at the cost of readability.

[image]

caquino, 2019-10-08:
@dipsylala which display are you using?

dipsylala, 2019-10-09:
I knew there was something I missed! InkyPhat, so 212x104.
I'm assuming UI testing's been on the 250x122 WS2, but the Slack channel has a lot of people mentioning the InkyPhat so maybe it needs to be considered too?

Could the UI assume 4 lines maximum for the Info area and use a combination of font scaling/line spacing to get it to fit depending on screen size?

evilsocket, 2019-10-09:
or maybe (as suggested by @hexwaxwing) the name could be on top of the face?

dipsylala, 2019-10-09:
I wasn't sure whether that might make it too crowded around the face - couple of mockups below. Shifting the name left and the text up certainly seems the quickest to implement.

WS2:
[image]

Inkyphat:
[image]

evilsocket, 2019-10-09:
i've tested on a waveshare v1, anybody can pull and test the fix for inky and papirus too?

dipsylala, 2019-10-09:
Looks great on inky - thanks!

[image]

evilsocket, 2019-10-09:
👍

closed 2019-10-09 by evilsocket

jonspraggins, 2021-02-05:
Similar issue.  When using PiSugar2 plugin and bt-tether, there's some crowding.  No option to hide bt-tether (but keep enabled) from what I can tell.
[image]