evilsocket/pwnagotchi #234: Missing dependencies for PaPiRus display

drego85, 2019-10-09:
**Describe the bug**
In v.1.0.0RC1 version there are missing drivers and some packages to make PaPiRus display work.

**Environment (please complete the following information):**
 -  v.1.0.0RC1
 - Raspberry Pi Zero W + PaPiRus 2"

**Additional context**

At the first start this error appears:

```
pi@pwnagotchi:/usr/local/bin $ sudo python3 pwnagotchi
[2019-07-10 01:57:56,928] [INFO] grid plugin loaded.
[2019-07-10 01:57:57,013] [WARNING] ui.fps is 0, the display will only update for major changes
[2019-07-10 01:57:57,016] [INFO] initializing papirus display
Traceback (most recent call last):
  File "pwnagotchi", line 38, in <module>
    display = Display(config=config, state={'name': '%s>' % pwnagotchi.name()})
  File "/usr/local/lib/python3.7/dist-packages/pwnagotchi/ui/display.py", line 104, in __init__
    self._init_display()
  File "/usr/local/lib/python3.7/dist-packages/pwnagotchi/ui/display.py", line 145, in _init_display
    from pwnagotchi.ui.papirus.epd import EPD
  File "/usr/local/lib/python3.7/dist-packages/pwnagotchi/ui/papirus/epd.py", line 18, in <module>
    from pwnagotchi.ui.papirus.lm75b import LM75B
  File "/usr/local/lib/python3.7/dist-packages/pwnagotchi/ui/papirus/lm75b.py", line 9, in <module>
    import smbus
ModuleNotFoundError: No module named 'smbus'
```

To solve, I did:
`sudo apt install python3-smbus`

Now I have this error:

```
pi@pwnagotchi:/usr/local/bin $ sudo python3 pwnagotchi
[2019-07-10 02:05:06,183] [INFO] grid plugin loaded.
[2019-07-10 02:05:06,270] [WARNING] ui.fps is 0, the display will only update for major changes
[2019-07-10 02:05:06,274] [INFO] initializing papirus display
Traceback (most recent call last):
  File "pwnagotchi", line 38, in <module>
    display = Display(config=config, state={'name': '%s>' % pwnagotchi.name()})
  File "/usr/local/lib/python3.7/dist-packages/pwnagotchi/ui/display.py", line 104, in __init__
    self._init_display()
  File "/usr/local/lib/python3.7/dist-packages/pwnagotchi/ui/display.py", line 147, in _init_display
    self._display = EPD()
  File "/usr/local/lib/python3.7/dist-packages/pwnagotchi/ui/papirus/epd.py", line 65, in __init__
    self._lm75b = LM75B()
  File "/usr/local/lib/python3.7/dist-packages/pwnagotchi/ui/papirus/lm75b.py", line 23, in __init__
    self._bus = smbus.SMBus(busnum)
FileNotFoundError: [Errno 2] No such file or directory
```

To solve, I enabled SPI and I2C:

```
sudo raspi-config nonint do_spi 0
sudo raspi-config nonint do_i2c 0
```

Now I have this error:

```
pi@pwnagotchi:/usr/local/bin $ sudo python3 pwnagotchi
[2019-07-10 02:06:46,064] [INFO] grid plugin loaded.
[2019-07-10 02:06:46,150] [WARNING] ui.fps is 0, the display will only update for major changes
[2019-07-10 02:06:46,154] [INFO] initializing papirus display
Traceback (most recent call last):
  File "pwnagotchi", line 38, in <module>
    display = Display(config=config, state={'name': '%s>' % pwnagotchi.name()})
  File "/usr/local/lib/python3.7/dist-packages/pwnagotchi/ui/display.py", line 104, in __init__
    self._init_display()
  File "/usr/local/lib/python3.7/dist-packages/pwnagotchi/ui/display.py", line 147, in _init_display
    self._display = EPD()
  File "/usr/local/lib/python3.7/dist-packages/pwnagotchi/ui/papirus/epd.py", line 83, in __init__
    with open(os.path.join(self._epd_path, 'version')) as f:
FileNotFoundError: [Errno 2] No such file or directory: '/dev/epd/version'
```

To solve, I did:
```

sudo apt install libfuse-dev bc fonts-freefont-ttf

git clone https://github.com/repaper/gratis.git

cd gratis
sudo make rpi EPD_IO=epd_io.h PANEL_VERSION='V231_G2'
sudo make rpi-install EPD_IO=epd_io.h PANEL_VERSION='V231_G2'

sudo systemctl enable epd-fuse.service

```

Finally you need to indicate the size of the display (there are several models):

```
sudo systemctl stop epd-fuse
sudo nano  /etc/default/epd-fuse

Edit line "#EPD_SIZE=2.0" with correct size  and remove "#"

sudo systemctl start epd-fuse
```
Restart and everything works.

Alternatively, one click solution:

`curl -sSL https://pisupp.ly/papiruscode | sudo bash`

evilsocket, 2019-10-09:
@caquino

caquino, 2019-10-09:
Working on it!

evilsocket, 2019-10-09:
fixed.

commit a93348db (2019-10-09): fix: installing missing smbus dependency (fixes #234)
closed 2019-10-09 by evilsocket