evilsocket/pwnagotchi #235: [BUG] waveshare v1 does not display pwnagotchi correctly

deveth0, 2019-10-09:
Since the installation of the RC1 my pwnagotchi does not display correctly anymore. 
I think one reason is, that the display is no longer reset when the gotchi starts (or at least it looks like this). 

**Describe the bug**
See imagd

**To Reproduce**
Steps to reproduce the behavior:
1. Install RC1 on a pi with waveshare v1 display
2. Enjoy

**Expected behavior**
I'd expect a clearer picture. 

**Screenshots**

[image]

**Environment (please complete the following information):**
 - Pwnagotchi version - RC1 
 - OS version
 - Type of hardware - pi0w, waveshare v1
 - Any additional hardware used

**Additional context**
Add any other context about the problem here.

evilsocket, 2019-10-09:
weird, it is cleared https://github.com/evilsocket/pwnagotchi/blob/master/pwnagotchi/ui/display.py#L156 ... might just be that the waveshare v1 have some pixels burnout problems?

deveth0, 2019-10-09:
Sry missed to say: with an older image it works fine (build a 2-3 days ago)

evilsocket, 2019-10-09:
even weirder, literally nothing in the v1 library code or in how it's used in the display.py changed from the first import: https://github.com/evilsocket/pwnagotchi/commit/cbf7511e1754a41a7f3144d1813603b0bb8b9cef#diff-358ffb3e7079891a9cb715e6738fba48

deveth0, 2019-10-09:
Yeah I also checked for changes. I'll get another pi with a waveshare tomorrow hopefully, will try, if I can reproduce it there. 

Currently it looks a bit as if the clear is canceled somehow. In older versions the complete screen turned black from outside to inside, now only a few pixel are black and then the first image is displayed.

evilsocket, 2019-10-09:
you can try to manually stop the service (`sudo service pwnagotchi stop`), clearing the display manually (`sudo pwnagotchi --clear`) and restart (`sudo service pwnagotchi start`).

It might be an hardware issue and the reason why there's a V2 of that display :D Your help debugging this is precious and appreciated, thanks 👍

hmax42, 2019-10-09:
i can confirm that behavior, except that i had the default waveshare logo on screen. then changed the setting to waveshare_v1 and rebootet. the image got darker and degraded, for the lack of a better word.

evilsocket, 2019-10-09:
does running a manual clear as described fixes the problem?

hmax42, 2019-10-09:
yes and no :)
funny thing is: i have an ethernet dongle which stayed connected during the advised power down, i just removed the wall wart from the wall.
this time i removed it, and it cleared the screen during boot/start.
then i logged in - before the face appeared - and stopped the service
the clear screen command worked.
service restart produced another clear screen then the face came on.

so it works

hmax42, 2019-10-09:
although, it didn't work on an usb connect to windows before...

deveth0, 2019-10-09:
i just tried to manually clear the screen using the command mentioned above. it just clears a few pixel on the outside and not the complete screen.

evilsocket, 2019-10-09:
it sounds like an hardware bug because that's a full clear

deveth0, 2019-10-09:
Damn, seems as if you are right and the display just broke down. I got a complete refresh using the waveshare sdk/examples but now not even this works anymore. Sad, just survived a week or so. I think this can be closed then. Can you recommend another display or was this just bad luck?

closed 2019-10-09 by deveth0

evilsocket, 2019-10-09:
waveshare v2