evilsocket/pwnagotchi #245: [BUG] Release RC2 SSH Login Message

gh0stshell, 2019-10-09:
**Describe the bug**
When logging in via SSH the login message appears and looks like this

Linux pwnagotchi 4.19.66-Re4son+ #1 Sun Aug 18 13:42:02 AEST 2019 armv6l
(■■■■) pwnagotchi (pwnagotchi-v1.0.0RC2)

**To Reproduce**
Steps to reproduce the behavior:
1. Download and flash RC2 image
2. Install into PiZeroW and connect to PC
3. Setup network and login via SSH

**Expected behavior**
I am guessing it is suppose to look like this:

Linux pwnagotchi 4.19.66-Re4son+ #1 Sun Aug 18 13:42:02 AEST 2019 armv6l
(◕‿‿◕) pwnagotchi (pwnagotchi-v1.0.0RC2)

**Screenshots**
https://ibb.co/bb0QzP4

**Environment (please complete the following information):**
 - Pwnagotchi 1.0 RC2
 - OS used to build the image
 - Pi Zero W
 - USB to micro USB cable

**Additional context**
N/A

leon-th, 2019-10-10:
This is not a problem of Pwnagotchi, it is a problem of Putty. It works on other SSH clients.
Leon T

gh0stshell, 2019-10-10:
Thats what I thought, just wanted to confirm

closed 2019-10-10 by gh0stshell