evilsocket/pwnagotchi #249: Option to share handshakes across peers

ellwoodthewood, 2019-10-10:
Since multiple pwnagotchi's split up the spectrum to scan, would it be possible to have a configurable true/false option for sharing the captured handshakes between them?

I.e. If three are together, in an area, and all have the "enable handshake sharing" set to true, over time, they should all have the same number of captured handshakes.

Option should be 'false' by default in case you don't want to share.

evilsocket, 2019-10-11:
what's the functional advantage of this (over, for instance, copying the handshakes yourself if needed...)?

ellwoodthewood, 2019-10-11:
The idea came from a meeting where three of us had pwnagotchi's and all had different pwnd numbers. We thought since they divide up the channels, they could share what was found on them with each other. This is assuming there are others you don't own, and want to share. Just a thought we discussed. It should take into account any blacklisting/do not share networks.

evilsocket, 2019-10-11:
with the advertising protocol would be very hard to transmit big chunks of data, **however**,  more units that pinned each other as friends (maybe by writing their fingerprints on the config?), might sync via pwngrid?

evilsocket, 2019-10-11:
on pwngrid everything is rsa encrypted so the server would only see encrypted data

Mal2012, 2019-10-31:
it should be possible to share handshakes via pwnmail. 

someone could create a plugin that sends handshakes via pwnmail when meeting a peer. the plugin needs to check the mail regulary and check if the recieved mail is a handshake. 

if it's a handshake copy it to /root/handshakes and delete mail. if not mark mail as unread again. 

similar to reported_handshakes it should create a white/blacklist to prevent duplicates 

when i have time i'll hack something together