evilsocket/pwnagotchi #250: [BUG] pwnagotchi stops working after short time

bitwave, 2019-10-10:
**Describe the bug**
After starting the pwnagotchi (RPI0W with waveshare v2) via PWR USB and waiting for around 5-10 min, the pwnagotchi stops working, no screen refreshes anymore, no ethernet ssh access. Same happens if started manually via ssh.

**To Reproduce**
Steps to reproduce the behavior:
1. Unplug pwnagotchi
2. Plug in (USB PWR)
3. wait < 30 min
4. no screen refreshes, looks dead

**Expected behavior**
The unit should stay on and refreshes the screen periodically, especially if it encounters new wifis.

**Screenshots**
None

**Environment (please complete the following information):**
 - Pwnagotchi version: RC2
 - Type of hardware: RPI0W
 - Any additional hardware used: waveshare v2 eink display

Leajian, 2019-10-10:
I had this problem before with a different project of mine, but not with pwnagotchi (I don't have a screen yet).
I suspect that the wifi enters power saving mode. So, try adding this line :
`iw dev mon0 set power_save off` in `/usr/bin/monstart` script and reboot.
Then, please report if this occurs again.

bitwave, 2019-10-10:
I tried it. And your fix saved the kernel, but not the python script:

```
[2019-10-10 21:39:51,679] [DEBUG] Starting new HTTP connection (1): localhost:8081
[2019-10-10 21:39:51,704] [DEBUG] Starting new HTTP connection (1): localhost:8081
[2019-10-10 21:39:51,839] [DEBUG] http://localhost:8081 "GET /api/events HTTP/1.1" 200 None
[2019-10-10 21:39:51,858] [DEBUG] http://localhost:8081 "POST /api/session HTTP/1.1" 200 26
[2019-10-10 21:39:51,982] [WARNING] !!! captured new handshake on channel 1: d8:55:75:64:b8:df () -> TC-91EFA [54:67:51:21:b5:3d (Compal Broadband Networks, Inc.)] !!!
[2019-10-10 21:40:05,608] [INFO] sending association frame to Schumann-Netzwerke (ec:23:3d:cf:40:9e Huawei Technologies Co.,Ltd) on channel 1 [0 clients]...
[2019-10-10 21:40:05,714] [DEBUG] Starting new HTTP connection (1): localhost:8081
[2019-10-10 21:40:05,852] [DEBUG] http://localhost:8081 "POST /api/session HTTP/1.1" 200 26
[2019-10-10 21:40:06,725] [DEBUG] Starting new HTTP connection (1): localhost:8081
[2019-10-10 21:40:08,233] [DEBUG] http://localhost:8081 "GET /api/session HTTP/1.1" 200 None
[2019-10-10 21:40:12,510] [INFO] waiting for 10s on channel 1 ...
[2019-10-10 21:40:12,589] [DEBUG] Starting new HTTP connection (1): localhost:8081
[2019-10-10 21:40:12,756] [DEBUG] http://localhost:8081 "GET /api/events HTTP/1.1" 200 None
[gotchialpha:00610] *** Process received signal ***
[gotchialpha:00610] Signal: Segmentation fault (11)
[gotchialpha:00610] Signal code: Address not mapped (1)
[gotchialpha:00610] Failing at address: 0x1
[gotchialpha:00610] *** End of error message ***
Segmentation fault
```

bitwave, 2019-10-10:
I found this: https://github.com/raspberrypi/linux/issues/2794

bitwave, 2019-10-11:
Looks like I fixed it with different power cable and more amps. Or maybe the SD-Card is shit. So if anyone has the same problem, change either the sd card or the cable/adapter or both

bitwave, 2019-10-11:
This is getting mysterious. On my laptop booted in manual mode, service stopped and started with `/usr/local/bin/pwnagotchi --debug` directly, it works, loads AI and ran for more than 1h. If i boot it with power only it get stuck after 5-10 min...

darumaseye, 2019-10-11:
I have had similar problems: in my case the contents of the card were corrupted and upon restart (unplug + plug) the system was not loaded. @evilsocket and others have suggested that the cause could be an old / bad microsd, in fact it seems to be a common problem with the microsd cards used as secondary memory, due to frequent writing, as mentioned here: https://www.raspberrypi.org/forums/viewtopic.php?t=36533
In my case I will try to buy a new micro sd, probably one of those with Application Performance class.. https://www.sdcard.org/developers/overview/application/index.html

I hope I've helped

bitwave, 2019-10-12:
I can load the system, the MANUAL mode works fine and longer than 10 min. The problem is the AUTO mode.

bitwave, 2019-10-12:
I already switched sd cards, same behavior for both.

bitwave, 2019-10-12:
I solved it using `over_voltage=1` in the config.txt

Steps to solve:
1. Check power cable and power supply
2. Check SD card
3. use the above fix, see https://github.com/raspberrypi/linux/issues/2555#issuecomment-539514732 for reference

Some models of the RPi0W have a H. on one of the coils, maybe these have problems.

closed 2019-10-12 by bitwave

domenukk, 2019-11-25:
This should probably go in the docs somewhere?

bitwave, 2019-11-26:
If you had the same problem & solved it for you, probably yes.