evilsocket/pwnagotchi #258: [BUG] Tweepy 3.6 is incompatible with python 3.7

dadav, 2019-10-11:
**Describe the bug**
Tweepy 3.6.0 doesnt support python 3.7.

**To Reproduce**
python3 -c 'import tweepy'

**Expected behavior**
Import is working

**Screenshots**
-

**Environment (please complete the following information):**
 - RC2

**Additional context**
Version 3.7.0 fixes the problem.

dadav, 2019-10-11:
Problem is: They use the async keyword as a variable name >.>

evilsocket, 2019-10-11:
uhm?
[image]

dadav, 2019-10-11:
hooowww??
Bug is definetly in 3.6.0:
https://github.com/tweepy/tweepy/blob/v3.6.0/tweepy/streaming.py#L358

It got fixed here: https://github.com/tweepy/tweepy/pull/1042
which is available in 3.7.0

gonna investigate!

dadav, 2019-10-13:
hm, maybe i upgraded the package while compiling the package on the pi somehow^^ Problem doesnt seem to occure to other ppl.

closed 2019-10-20 by evilsocket