evilsocket/pwnagotchi #262: [BUG] v1.0.0RC2 Image Kernel Panic

gaterbyte, 2019-10-11:
**Describe the bug**
v1.0.0RC2 Image Flash Results in a Kernel Panic

**To Reproduce**
Steps to reproduce the behavior:
1. Download image file
2. use Etcher to flash
3. insert micro sdcard into pi and power off

**Expected behavior**
When Reverting back to "pwnagotchi-raspbian-lite-master.img" pi boots normally.

**Screenshots**
[image]

**Environment (please complete the following information):**
 - pwnagotchi-raspbian-lite-v1.0.0RC2
 - Pi 0 W
 - 32GB Pny SD card, inateck usb/ethernet hub, mini-hdmi to monitor

evilsocket, 2019-10-11:
does the release SHA256 match with your zip?

gaterbyte, 2019-10-11:
cd498ed144f464068dcab6f7648460803df242baa8a0f940054094035ffcfb52  
pwnagotchi-raspbian-lite-v1.0.0RC2.zip

f8d6ad92a4d0d0760850330379d1254c9e4d2ba8f4841d2563b1032ab4eb505b  
pwnagotchi-raspbian-lite-v1.0.0RC2.img

Which are different from what you posted... Let me try to re-download and report back.

gaterbyte, 2019-10-11:
Pull via wget verses chrome download, and sha256 now matches.
Checking install and running now.

closed 2019-10-11 by gaterbyte