evilsocket/pwnagotchi #267: document nexmon blindness bug

evilsocket, 2019-10-12:
every once in a while, nexmon dies with:

```
[ 4341.527847] brcmfmac: brcmf_cfg80211_nexmon_set_channel: Set Channel failed: chspec=4101, -110
[ 4344.327806] brcmfmac: brcmf_cfg80211_nexmon_set_channel: Set Channel failed: chspec=4097, -110
[ 4347.127853] brcmfmac: brcmf_cfg80211_nexmon_set_channel: Set Channel failed: chspec=4098, -110
[ 4349.927917] brcmfmac: brcmf_cfg80211_nexmon_set_channel: Set Channel failed: chspec=4099, -110
[ 4352.728074] brcmfmac: brcmf_cfg80211_nexmon_set_channel: Set Channel failed: chspec=4100, -110
[ 4355.527970] brcmfmac: brcmf_cfg80211_nexmon_set_channel: Set Channel failed: chspec=4101, -110
[ 4358.328022] brcmfmac: brcmf_cfg80211_nexmon_set_channel: Set Channel failed: chspec=4102, -110
[ 4361.208095] brcmfmac: brcmf_cfg80211_nexmon_set_channel: Set Channel failed: chspec=4103, -110
[ 4364.008157] brcmfmac: brcmf_cfg80211_nexmon_set_channel: Set Channel failed: chspec=4104, -110
[ 4366.808218] brcmfmac: brcmf_cfg80211_nexmon_set_channel: Set Channel failed: chspec=4105, -110
[ 4369.608431] brcmfmac: brcmf_cfg80211_nexmon_set_channel: Set Channel failed: chspec=4097, -110
[ 4372.408345] brcmfmac: brcmf_cfg80211_nexmon_set_channel: Set Channel failed: chspec=4098, -110
[ 4375.288408] brcmfmac: brcmf_cfg80211_nexmon_set_channel: Set Channel failed: chspec=4099, -110
[ 4378.088474] brcmfmac: brcmf_cfg80211_nexmon_set_channel: Set Channel failed: chspec=4100, -110
[ 4380.891399] brcmfmac: brcmf_cfg80211_nexmon_set_channel: Set Channel failed: chspec=4101, -110
```

And only a reboot can fix the wifi, this is why the `mon_max_blind_epochs` parameter exists, to reboot the board when this happens.

Ideally we should document this known issue, the configuration and some day maybe fix it.

evilsocket, 2019-10-12:
cc @hexwaxwing

evilsocket, 2019-10-12:
docs added https://pwnagotchi.ai/usage/#known-issues

Leajian, 2019-10-13:
You could try reloading the driver instead of rebooting the pi, using :
```
modprobe -r brcmfmac
modprobe brcmfmac
```
If this still fails, then a reboot will fix this.
There is a great discussion about that problem here : https://www.bountysource.com/issues/56252669-wlan-freezes-in-raspberry-pi-3b

evilsocket, 2019-10-13:
tried that way, it doesn't always work, the only reliable way is rebooting

JRWR, 2019-10-14:
So, The Nexmon firmware is a little picky on how its interfaced, How are your bringing up the mon interface in linux before bettercap gets to it?

evilsocket, 2019-10-14:
iw phy phy0 interface add mon0 type monitor && ifconfig mon0 up

evilsocket, 2019-10-14:
from https://github.com/evilsocket/pwnagotchi/blob/master/builder/pwnagotchi.yml#L333

evilsocket, 2019-10-14:
(from re4son monstart script)

JRWR, 2019-10-15:
Ok, So I looked, We are using the older version of the firmware provided by Nexmon, I think Re4son kernel does it as its the default for Nexmon to use that. Patches and Commits from the Nexmon Project show 7.45.189 as the latest version you can use (The base firmware comes from the OEM, not the RPI foundation) as of RC4, we are using version 7.45.154 of the bcm43455c0 firmware This issue should only affect 3B+ and 4s, the 3B and the 0W use the same wifi chip and are listed as supporting injection.

JRWR, 2019-10-15:
Good News, The Nexmon Patches with the stock kernel works well, Here is my DMesg output after running for 10 minutes, I will be running it for the next 24 hours to see if its stable, 

`
[    5.128852] brcmfmac: loading out-of-tree module taints kernel.
[    5.128864] brcmfmac: loading out-of-tree module taints kernel.
[    5.181306] brcmfmac: F1 signature read @0x18000000=0x15264345
[    5.190412] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[    5.190907] usbcore: registered new interface driver brcmfmac
[    5.566003] brcmfmac: brcmf_sdio_bus_preinit: before brcmf_sdio_debugfs_create
[    5.569776] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[    5.597271] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/6 wl0: Oct 15 2019 20:30:25 version 7.45.189 (nexmon.org: -4) FWID 01-e1db26e2
[    5.675296] brcmfmac: brcmf_bus_started: before brcmf_debugfs_add_entry
[    8.495009] brcmfmac: brcmf_vif_add_validate: Attempt to add a MONITOR interface...
[    8.495024] brcmfmac: brcmf_mon_add_vif: brcmf_mon_add_vif called
[    8.495028] brcmfmac: brcmf_mon_add_vif: Adding vif "mon0"

`

JRWR, 2019-10-15:
Good news, It's stable ! 
[image]

JRWR, 2019-10-15:
And I failed, You get a little more debug output this time overall..

```
[ 5387.511962] brcmfmac: brcmf_cfg80211_nexmon_set_channel: Set Channel failed: chspec=4097, -110
[ 5390.551934] brcmfmac: brcmf_cfg80211_nexmon_set_channel: Set Channel failed: chspec=4098, -110
[ 5393.601927] brcmfmac: brcmf_cfg80211_nexmon_set_channel: Set Channel failed: chspec=4099, -110
[ 5396.631949] brcmfmac: brcmf_cfg80211_nexmon_set_channel: Set Channel failed: chspec=4101, -110
[ 5399.671984] brcmfmac: brcmf_cfg80211_nexmon_set_channel: Set Channel failed: chspec=4102, -110
[ 5402.711975] brcmfmac: brcmf_cfg80211_nexmon_set_channel: Set Channel failed: chspec=4103, -110
[ 5405.752020] brcmfmac: brcmf_cfg80211_nexmon_set_channel: Set Channel failed: chspec=4104, -110
[ 5408.791991] brcmfmac: brcmf_cfg80211_nexmon_set_channel: Set Channel failed: chspec=4105, -110
[ 5411.831985] brcmfmac: brcmf_cfg80211_nexmon_set_channel: Set Channel failed: chspec=4106, -110
[ 5414.872013] brcmfmac: brcmf_cfg80211_nexmon_set_channel: Set Channel failed: chspec=53284, -110
[ 5417.911966] brcmfmac: brcmf_cfg80211_nexmon_set_channel: Set Channel failed: chspec=53288, -110
[ 5420.951985] brcmfmac: brcmf_cfg80211_nexmon_set_channel: Set Channel failed: chspec=53296, -110
[ 5423.511967] brcmfmac: brcmf_cfg80211_nexmon_set_channel: Set Channel failed: chspec=4099, -110
[ 5426.312006] brcmfmac: brcmf_cfg80211_nexmon_set_channel: Set Channel failed: chspec=4100, -110
[ 5429.111938] brcmfmac: brcmf_cfg80211_nexmon_set_channel: Set Channel failed: chspec=4103, -110
[ 5431.911970] brcmfmac: brcmf_cfg80211_nexmon_set_channel: Set Channel failed: chspec=4105, -110
[ 5434.711974] brcmfmac: brcmf_cfg80211_nexmon_set_channel: Set Channel failed: chspec=53308, -110
[ 5437.511963] brcmfmac: brcmf_cfg80211_nexmon_set_channel: Set Channel failed: chspec=53348, -110
[ 5440.311927] brcmfmac: brcmf_cfg80211_nexmon_set_channel: Set Channel failed: chspec=53356, -110
[ 5443.111922] brcmfmac: brcmf_cfg80211_nexmon_set_channel: Set Channel failed: chspec=53360, -110
[ 5445.911981] brcmfmac: brcmf_cfg80211_nexmon_set_channel: Set Channel failed: chspec=53368, -110
[ 5448.711973] brcmfmac: brcmf_cfg80211_nexmon_set_channel: Set Channel failed: chspec=53372, -110
[ 5451.511925] brcmfmac: brcmf_cfg80211_nexmon_set_channel: Set Channel failed: chspec=53380, -110
[ 5454.311928] brcmfmac: brcmf_cfg80211_nexmon_set_channel: Set Channel failed: chspec=4099, -110
[ 5457.111929] brcmfmac: brcmf_cfg80211_nexmon_set_channel: Set Channel failed: chspec=4100, -110
[ 5459.911983] brcmfmac: brcmf_cfg80211_nexmon_set_channel: Set Channel failed: chspec=4103, -110
[ 5462.711974] brcmfmac: brcmf_cfg80211_nexmon_set_channel: Set Channel failed: chspec=4105, -110
[ 5465.511924] brcmfmac: brcmf_cfg80211_nexmon_set_channel: Set Channel failed: chspec=53308, -110
[ 5468.311926] brcmfmac: brcmf_cfg80211_nexmon_set_channel: Set Channel failed: chspec=53348, -110
[ 5471.111925] brcmfmac: brcmf_cfg80211_nexmon_set_channel: Set Channel failed: chspec=53356, -110
[ 5473.912031] brcmfmac: brcmf_cfg80211_nexmon_set_channel: Set Channel failed: chspec=53360, -110
[ 5476.471932] brcmfmac: brcmf_cfg80211_nexmon_set_channel: Set Channel failed: chspec=4102, -110
[ 5479.271932] brcmfmac: brcmf_cfg80211_nexmon_set_channel: Set Channel failed: chspec=4103, -110
[ 5482.081987] brcmfmac: brcmf_cfg80211_nexmon_set_channel: Set Channel failed: chspec=4104, -110
[ 5484.871994] brcmfmac: brcmf_cfg80211_nexmon_set_channel: Set Channel failed: chspec=4105, -110
[ 5487.671927] brcmfmac: brcmf_cfg80211_nexmon_set_channel: Set Channel failed: chspec=53292, -110
[ 5490.471930] brcmfmac: brcmf_cfg80211_nexmon_set_channel: Set Channel failed: chspec=53300, -110
[ 5493.271977] brcmfmac: brcmf_cfg80211_nexmon_set_channel: Set Channel failed: chspec=53304, -110
[ 5496.071923] brcmfmac: brcmf_cfg80211_nexmon_set_channel: Set Channel failed: chspec=53348, -110
[ 5498.872004] brcmfmac: brcmf_cfg80211_nexmon_set_channel: Set Channel failed: chspec=53364, -110
[ 5501.671976] brcmfmac: brcmf_cfg80211_nexmon_set_channel: Set Channel failed: chspec=53372, -110
[ 5504.471928] brcmfmac: brcmf_cfg80211_nexmon_set_channel: Set Channel failed: chspec=53376, -110
[ 5507.031989] brcmfmac: brcmf_proto_bcdc_query_dcmd: brcmf_proto_bcdc_msg failed w/status -110
[ 5507.032002] brcmfmac: brcmf_cfg80211_get_channel: chanspec failed (-110)
[ 5509.591930] brcmfmac: brcmf_proto_bcdc_query_dcmd: brcmf_proto_bcdc_msg failed w/status -110
[ 5509.591939] brcmfmac: brcmf_cfg80211_get_tx_power: error (-110)
[ 5512.151989] brcmfmac: brcmf_cfg80211_nexmon_set_channel: Set Channel failed: chspec=4102, -110
[ 5514.952027] brcmfmac: brcmf_cfg80211_nexmon_set_channel: Set Channel failed: chspec=4103, -110

```

evilsocket, 2019-10-16:
yep when that happens even trying to change channel manually doesn't work, i think it's the heat

JRWR, 2019-10-16:
Based off this photo, There is no TIM under the wifi can at all... Since its used for shielding RF, this is of no surprise. Someone should get a IR Camera on the board and take a look.  

 [image]

JRWR, 2019-10-16:
So, I'm looking at the datasheet, for the CYW43455 It states that the Max temp for operation is 120C while under normal loads on a 4 layer board Of course Max oper temp is +85C The thing only puts out 1.2W but I did notice that on the older RPI0W Chips it had a self limiter for overheat, This one is not stated in the data sheet as having one

evilsocket, 2019-10-17:
so you're saying that it's the chinese sabotaging our wifi pwning, right?

JRWR, 2019-10-17:
Maybe.... Until I can get a proper Temp readout of the die itself while under the heavy load, We wont know.

evilsocket, 2019-10-17:
_suspence music_

DrSchottky, 2019-10-20:
Hi,
is the same issue afflicting both 43455c0 and 43430a1?
Have you got a reliable way to reproduce it?

evilsocket, 2019-11-01:
seems related

```
[ 2903.917746] WARNING: CPU: 0 PID: 4695 at /opt/kernel-builder_repos/Re4son/re4son-raspberrypi-linux/all/fs/sysfs/group.c:255 sysfs_remove_group+0x98/0xa4
[ 2903.917753] sysfs group 'byte_queue_limits' not found for kobject 'tx-0'
[ 2903.917757] Modules linked in: evdev uinput bnep hci_uart btbcm serdev bluetooth ecdh_generic fuse spidev brcmfmac brcmutil raspberrypi_hwmon snd_bcm2835(C) hwmon cfg80211 snd_pcm snd_timer rfkill snd bcm2835_codec(C) v4l2_mem2mem bcm2835_v4l2(C) i2c_bcm2835 bcm2835_mmal_vchiq(C) spi_bcm2835aux v4l2_common videobuf2_dma_contig videobuf2_vmalloc videobuf2_memops videobuf2_v4l2 videobuf2_common spi_bcm2835 videodev media vc_sm_cma(C) fixed uio_pdrv_genirq uio i2c_dev usb_f_ecm g_ether usb_f_rndis u_ether libcomposite dwc2 udc_core ip_tables x_tables ipv6
[ 2903.917930] CPU: 0 PID: 4695 Comm: kbnepd bnep0 Tainted: G         C        4.19.66-Re4son+ #1
[ 2903.917935] Hardware name: BCM2835
[ 2903.917969] [<c0017e44>] (unwind_backtrace) from [<c0014d84>] (show_stack+0x20/0x24)
[ 2903.917992] [<c0014d84>] (show_stack) from [<c06f3278>] (dump_stack+0x20/0x28)
[ 2903.918075] [<c06f3278>] (dump_stack) from [<c0024624>] (__warn+0xf0/0x118)
[ 2903.918092] [<c0024624>] (__warn) from [<c002469c>] (warn_slowpath_fmt+0x50/0x70)
[ 2903.918112] [<c002469c>] (warn_slowpath_fmt) from [<c021f370>] (sysfs_remove_group+0x98/0xa4)
[ 2903.918131] [<c021f370>] (sysfs_remove_group) from [<c0613460>] (netdev_queue_update_kobjects+0xc0/0x178)
[ 2903.918146] [<c0613460>] (netdev_queue_update_kobjects) from [<c0613578>] (netdev_unregister_kobject+0x60/0x80)
[ 2903.918169] [<c0613578>] (netdev_unregister_kobject) from [<c05eb0f4>] (rollback_registered_many+0x1ec/0x508)
[ 2903.918185] [<c05eb0f4>] (rollback_registered_many) from [<c05eb4ac>] (unregister_netdevice_queue+0x9c/0x130)
[ 2903.918199] [<c05eb4ac>] (unregister_netdevice_queue) from [<c05eb568>] (unregister_netdev+0x28/0x30)
[ 2903.918237] [<c05eb568>] (unregister_netdev) from [<bf5f1958>] (bnep_session+0x4d4/0x888 [bnep])
[ 2903.918288] [<bf5f1958>] (bnep_session [bnep]) from [<c00432c0>] (kthread+0x120/0x160)
[ 2903.918305] [<c00432c0>] (kthread) from [<c00090ac>] (ret_from_fork+0x14/0x28)
[ 2903.918312] Exception stack(0xd67affb0 to 0xd67afff8)
[ 2903.918321] ffa0:                                     00000000 00000000 00000000 00000000
[ 2903.918332] ffc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[ 2903.918340] ffe0: 00000000 00000000 00000000 00000000 00000013 00000000
[ 2903.918347] ---[ end trace cfb2c6995ec3c276 ]---
...
[ 2903.918482] WARNING: CPU: 0 PID: 4695 at /opt/kernel-builder_repos/Re4son/re4son-raspberrypi-linux/all/fs/sysfs/group.c:255 sysfs_remove_group+0x98/0xa4
[ 2903.918488] sysfs group 'power' not found for kobject 'bnep0'
[ 2903.918492] Modules linked in: evdev uinput bnep hci_uart btbcm serdev bluetooth ecdh_generic fuse spidev brcmfmac brcmutil raspberrypi_hwmon snd_bcm2835(C) hwmon cfg80211 snd_pcm snd_timer rfkill snd bcm2835_codec(C) v4l2_mem2mem bcm2835_v4l2(C) i2c_bcm2835 bcm2835_mmal_vchiq(C) spi_bcm2835aux v4l2_common videobuf2_dma_contig videobuf2_vmalloc videobuf2_memops videobuf2_v4l2 videobuf2_common spi_bcm2835 videodev media vc_sm_cma(C) fixed uio_pdrv_genirq uio i2c_dev usb_f_ecm g_ether usb_f_rndis u_ether libcomposite dwc2 udc_core ip_tables x_tables ipv6
[ 2903.918652] CPU: 0 PID: 4695 Comm: kbnepd bnep0 Tainted: G        WC        4.19.66-Re4son+ #1
[ 2903.918657] Hardware name: BCM2835
[ 2903.918682] [<c0017e44>] (unwind_backtrace) from [<c0014d84>] (show_stack+0x20/0x24)
[ 2903.918701] [<c0014d84>] (show_stack) from [<c06f3278>] (dump_stack+0x20/0x28)
[ 2903.918722] [<c06f3278>] (dump_stack) from [<c0024624>] (__warn+0xf0/0x118)
[ 2903.918738] [<c0024624>] (__warn) from [<c002469c>] (warn_slowpath_fmt+0x50/0x70)
[ 2903.918756] [<c002469c>] (warn_slowpath_fmt) from [<c021f370>] (sysfs_remove_group+0x98/0xa4)
[ 2903.918781] [<c021f370>] (sysfs_remove_group) from [<c04a1474>] (dpm_sysfs_remove+0x58/0x60)
[ 2903.918798] [<c04a1474>] (dpm_sysfs_remove) from [<c0494e48>] (device_del+0x7c/0x360)
[ 2903.918813] [<c0494e48>] (device_del) from [<c0613594>] (netdev_unregister_kobject+0x7c/0x80)
[ 2903.918832] [<c0613594>] (netdev_unregister_kobject) from [<c05eb0f4>] (rollback_registered_many+0x1ec/0x508)
[ 2903.918847] [<c05eb0f4>] (rollback_registered_many) from [<c05eb4ac>] (unregister_netdevice_queue+0x9c/0x130)
[ 2903.918861] [<c05eb4ac>] (unregister_netdevice_queue) from [<c05eb568>] (unregister_netdev+0x28/0x30)
[ 2903.918891] [<c05eb568>] (unregister_netdev) from [<bf5f1958>] (bnep_session+0x4d4/0x888 [bnep])
[ 2903.918919] [<bf5f1958>] (bnep_session [bnep]) from [<c00432c0>] (kthread+0x120/0x160)
[ 2903.918933] [<c00432c0>] (kthread) from [<c00090ac>] (ret_from_fork+0x14/0x28)
[ 2903.918940] Exception stack(0xd67affb0 to 0xd67afff8)
[ 2903.918949] ffa0:                                     00000000 00000000 00000000 00000000
[ 2903.918959] ffc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[ 2903.918968] ffe0: 00000000 00000000 00000000 00000000 00000013 00000000
[ 2903.918973] ---[ end trace cfb2c6995ec3c277 ]---
...
[ 2903.919015] WARNING: CPU: 0 PID: 4695 at /opt/kernel-builder_repos/Re4son/re4son-raspberrypi-linux/all/fs/sysfs/group.c:255 sysfs_remove_group+0x98/0xa4
[ 2903.919021] sysfs group 'statistics' not found for kobject 'bnep0'
[... cut ...]
[ 3631.199994] [<c01ae678>] (sys_ioctl) from [<c0009000>] (ret_fast_syscall+0x0/0x28)
[ 3631.199997] Exception stack(0xd8b27fa8 to 0xd8b27ff0)
[ 3631.200001] 7fa0:                   00000000 000f1828 0000000a 400442c8 bec1e458 00000001
[ 3631.200005] 7fc0: 00000000 000f1828 002af556 00000036 00000014 00000000 b6e59448 00000000
[ 3631.200008] 7fe0: 000f1c5c bec1e454 0003ba58 b6bb951c
[ 3631.200011] Code: 3a000019 e3530000 e243e001 0a00001b (e5d2c000) 
[ 3631.200014] ---[ end trace cfb2c6995ec3c27b ]---
```

hackabean, 2019-11-09:
@evilsocket 
hey, I am investigating "brcmf_cfg80211_nexmon_set_channel: Set Channel failed: chspec=4101, -110" aka Nexmon blindness bug issue and have noticed something that can potentially lead to resolution of the problem. My understanding was that patched Nexmon failed to set a channel, so instead of hopping on all channels I specified the channels manually to use for sending deauth frames in config.yml. In my case most AP are sitting on 1,4,6 and 11. Gotchi has been running for 15 minutes so far without any crashes. Can somebody confirm if it can be a valid solution ?

evilsocket, 2019-11-09:
@hackabean well not really, you just worked around the issue, but when that happens the whole wifi chip stops being very responsive

evilsocket, 2019-11-09:
if you specify the channels manually it will still hop ... i wonder if that's an issue related to hopping too quickly ...

evilsocket, 2019-11-09:
maybe increasing the [wifi.hop.period](https://www.bettercap.org/modules/wifi/)

hackabean, 2019-11-09:
@evilsocket I think it might be as simple as that I have increased hop_recon_time: from 10 to 30 and it stays error free. (my recon_time: 60 additionaly)

DrSchottky, 2019-11-09:
@hackabean -110 means firmware hanged.
When working on nexmon I experienced blindness followed by cmds timeout when I was leaking memory in rx queue. Timeout/traps are fw bugs.

hackabean, 2019-11-09:
@DrSchottky thanks for stepping in, seems like you know more about this issue. I just wished I could do something to make this less problematic so I was just poking around hoping to find a workaround. I am crossing my fingers that somebody will take a look at it.

DrSchottky, 2019-11-10:
@hackabean to work on the problem I need to know as much as possible about the setup (board and kernel/driver/fw versions), the environment (does it crash under particular conditions? How often?) and I need to have a reliable way to reproduce the bug on my testbed.
AFAIK monitor mode should be stable (never used injection in deep so far)

andrewbeard, 2019-11-11:
@DrSchottky I can reproduce it pretty reliably on my Pi4 setup. I may need a little help gathering the info you need, but I'm pretty confident it will go blind within an hour every time I power it up. What can I do to help?

DrSchottky, 2019-11-11:
@andrewbeard I need to know:

- Kernel: `uname -a`
- bcm fw: `dmesg | grep 7.45`

andrewbeard, 2019-11-11:
> @andrewbeard I need to know:
> 
> * Kernel: `uname -a`
`Linux pwnpi4 4.19.66-Re4son-v7l+ #1 SMP Mon Aug 19 08:12:03 AEST 2019 armv7l GNU/Linux`
> * bcm fw: `dmesg | grep 7.45`
'[    2.749080] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/6 wl0: Oct  8 2018 09:18:41 version 7.45.154 (nexmon.org: 2.2.2-195-g01e0-1) FWID 01-4fbe0b04'

andrewbeard, 2019-11-11:
System went blind within 3 minutes of powering up:

`[    2.739241] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[    2.749080] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/6 wl0: Oct  8 2018 09:18:41 version 7.45.154 (nexmon.org: 2.2.2-195-g01e0-1) FWID 01-4fbe0b04
[    3.185453] bcmgenet: Skipping UMAC reset
[    3.301924] bcmgenet fd580000.genet: configuring instance for external RGMII (no delay)
[    3.302331] IPv6: ADDRCONF(NETDEV_UP): eth0: link is not ready
[    3.692992] uart-pl011 fe201000.serial: no DMA platform data
[    3.826165] fuse init (API version 7.27)
[    3.892931] brcmfmac: brcmf_vif_add_validate: Attempt to add a MONITOR interface...
[    3.892940] brcmfmac: brcmf_mon_add_vif: brcmf_mon_add_vif called
[    3.892944] brcmfmac: brcmf_mon_add_vif: Adding vif "mon0"
[    4.222668] Adding 102396k swap on /var/swap.  Priority:-2 extents:2 across:106496k SSFS
[    4.311539] bcmgenet fd580000.genet eth0: Link is Down
[    8.471570] bcmgenet fd580000.genet eth0: Link is Up - 1Gbps/Full - flow control rx/tx
[    8.471593] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[   10.595212] Bluetooth: Core ver 2.22
[   10.595265] NET: Registered protocol family 31
[   10.595270] Bluetooth: HCI device and connection manager initialized
[   10.596016] Bluetooth: HCI socket layer initialized
[   10.596026] Bluetooth: L2CAP socket layer initialized
[   10.596069] Bluetooth: SCO socket layer initialized
[   10.604001] Bluetooth: HCI UART driver ver 2.3
[   10.604008] Bluetooth: HCI UART protocol H4 registered
[   10.604064] Bluetooth: HCI UART protocol Three-wire (H5) registered
[   10.604164] Bluetooth: HCI UART protocol Broadcom registered
[   10.763404] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[   10.763411] Bluetooth: BNEP filters: protocol multicast
[   10.763423] Bluetooth: BNEP socket layer initialized
[  156.774707] brcmfmac: brcmf_sdio_hostmail: mailbox indicates firmware halted
[  160.556985] brcmfmac: brcmf_cfg80211_nexmon_set_channel: Set Channel failed: chspec=4099, -110
[  163.116341] brcmfmac: brcmf_cfg80211_nexmon_set_channel: Set Channel failed: chspec=4107, -110
[  165.996384] brcmfmac: brcmf_cfg80211_nexmon_set_channel: Set Channel failed: chspec=4103, -110
[  168.796416] brcmfmac: brcmf_cfg80211_nexmon_set_channel: Set Channel failed: chspec=4104, -110
[  171.596449] brcmfmac: brcmf_cfg80211_nexmon_set_channel: Set Channel failed: chspec=4109, -110
[  174.396468] brcmfmac: brcmf_cfg80211_nexmon_set_channel: Set Channel failed: chspec=53284, -110
[  177.196499] brcmfmac: brcmf_cfg80211_nexmon_set_channel: Set Channel failed: chspec=53300, -110
[  179.996566] brcmfmac: brcmf_cfg80211_nexmon_set_channel: Set Channel failed: chspec=53304, -110
[  182.636544] brcmfmac: brcmf_cfg80211_nexmon_set_channel: Set Channel failed: chspec=4102, -110
[  209.916754] brcmfmac: brcmf_cfg80211_nexmon_set_channel: Set Channel failed: chspec=53284, -110
[  247.996974] brcmfmac: brcmf_cfg80211_nexmon_set_channel: Set Channel failed: chspec=4099, -110
[  255.597015] brcmfmac: brcmf_cfg80211_nexmon_set_channel: Set Channel failed: chspec=4104, -110
[  258.397045] brcmfmac: brcmf_cfg80211_nexmon_set_channel: Set Channel failed: chspec=4107, -110
[  261.197034] brcmfmac: brcmf_cfg80211_nexmon_set_channel: Set Channel failed: chspec=4108, -110`

DrSchottky, 2019-11-11:
7.45.154 is an outdated base fw (and if it cames form re4son-nexmon it also has outdated patches. Avoid it)
Broadcom's built in monitor mode for bcm43455c0(RPi 3B+/4) is broken, but [we fixed it ](https://github.com/seemoo-lab/nexmon/commit/17cc1980d27edd3440fbfe2022717675ba3ca8f2#diff-abbdc05d8a6c7c0ca2aced9affe6e9bc) (at least the bug we found) from nexmon in 7.45.189 patchset. You should get the headers for your kernel and build fw + driver with the latest patches from nexmon official repo.

I'm not telling you that it'll solve your problem, but before we go any further I need you to have the latest nexmon running on your system.

andrewbeard, 2019-11-11:
@DrSchottky I am totally willing to do that, but I may need a bit of hand holding as I don't have a dev environment for this system yet. So far I've just been writing images to SD cards and changing config files.  Any chance you (or anyone else) can point me at some info for building and replacing the kernel on a pi?

DrSchottky, 2019-11-11:
As far as I read the released pwnagotchi img includes the header package, so ssh into your device and start following [the README](https://github.com/seemoo-lab/nexmon#build-patches-for-bcm43430a1-on-the-rpi3zero-w-or-bcm434355c0-on-the-rpi3rpi4-using-raspbian-recommended) (but do not install the raspberrypi-kernel-headers package!)

andrewbeard, 2019-11-11:
I'm trying to run through that now, but I've run into a couple snags. Most of them I'm getting through (aclocal version mismatches, etc) but my end result didn't go all that well:

`  COLLECTING STATISTICS read /usr/src/nexmon/STATISTICS.md for more information
  COMPILING src/version.c => obj/version.o (details: log/compiler.log)
  GENERATING LINKER FILE gen/nexmon.pre => gen/nexmon.ld
  GENERATING LINKER FILE gen/nexmon.pre => gen/flashpatches.ld
  LINKING OBJECTS => gen/patch.elf (details: log/linker.log, log/linker.err)
  GENERATING MAKE FILE gen/nexmon.pre => gen/nexmon.mk
  GENERATING MAKE FILE gen/nexmon.pre => gen/flashpatches.mk
  APPLYING FLASHPATCHES gen/flashpatches.mk => brcmfmac43455-sdio.bin (details: log/flashpatches.log)
  APPLYING PATCHES gen/nexmon.mk => brcmfmac43455-sdio.bin (details: log/patches.log)
  BUILDING DRIVER for kernel 4.19 brcmfmac_4.19.y-nexmon/brcmfmac.ko (details: log/driver.log)
  COPYING brcmfmac43455-sdio.bin => /lib/firmware/brcm/brcmfmac43455-sdio.bin
  UNLOADING brcmfmac
  RELOADING brcmfmac
insmod: ERROR: could not insert module brcmfmac_4.19.y-nexmon/brcmfmac.ko: Invalid module format
make: *** [Makefile:192: install-firmware] Error 1
`

DrSchottky, 2019-11-11:
The module you're compiling is not targeting your running kernel.
Post the result of `modinfo brcmfmac_4.19.y-nexmon/brcmfmac.ko`

andrewbeard, 2019-11-11:
root@pwnpi4:/usr/src/nexmon/patches/bcm43455c0/7_45_189/nexmon# modinfo brcmfmac_4.19.y-nexmon/brcmfmac.ko
filename:       /usr/src/nexmon/patches/bcm43455c0/7_45_189/nexmon/brcmfmac_4.19.y-nexmon/brcmfmac.ko
license:        Dual BSD/GPL
description:    Broadcom 802.11 wireless LAN fullmac driver.
author:         Broadcom Corporation
firmware:       brcm/brcmfmac4373-sdio.bin
firmware:       brcm/brcmfmac4356-sdio.bin
firmware:       brcm/brcmfmac4354-sdio.bin
firmware:       brcm/brcmfmac43455-sdio.bin
firmware:       brcm/brcmfmac43430-sdio.bin
firmware:       brcm/brcmfmac43430a0-sdio.bin
firmware:       brcm/brcmfmac4339-sdio.bin
firmware:       brcm/brcmfmac43362-sdio.bin
firmware:       brcm/brcmfmac4335-sdio.bin
firmware:       brcm/brcmfmac43340-sdio.bin
firmware:       brcm/brcmfmac4334-sdio.bin
firmware:       brcm/brcmfmac4330-sdio.bin
firmware:       brcm/brcmfmac4329-sdio.bin
firmware:       brcm/brcmfmac43241b5-sdio.bin
firmware:       brcm/brcmfmac43241b4-sdio.bin
firmware:       brcm/brcmfmac43241b0-sdio.bin
firmware:       brcm/brcmfmac43143-sdio.bin
firmware:       brcm/brcmfmac4373.bin
firmware:       brcm/brcmfmac43569.bin
firmware:       brcm/brcmfmac43242a.bin
firmware:       brcm/brcmfmac43236b.bin
firmware:       brcm/brcmfmac43143.bin
srcversion:     A208FF46FF4FDC4A48B10E7
alias:          usb:v04B4p0BDCd*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0A5Cp0BDCd*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v043Ep3101d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v04B4pBD29d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v13B1p0039d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0A5CpBD27d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0A5CpBD1Fd*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0A5CpBD17d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0A5CpBD1Ed*dc*dsc*dp*ic*isc*ip*in*
alias:          sdio:c*v02D0d4373*
alias:          sdio:c*v02D0d4356*
alias:          sdio:c*v02D0d4354*
alias:          sdio:c*v02D0dA9BF*
alias:          sdio:c*v02D0d4345*
alias:          sdio:c*v02D0dA9A6*
alias:          sdio:c*v02D0d4339*
alias:          sdio:c*v02D0d4335*
alias:          sdio:c*v02D0dA9A4*
alias:          sdio:c*v02D0dA962*
alias:          sdio:c*v02D0dA94D*
alias:          sdio:c*v02D0dA94C*
alias:          sdio:c*v02D0d4334*
alias:          sdio:c*v02D0d4330*
alias:          sdio:c*v02D0d4329*
alias:          sdio:c*v02D0d4324*
alias:          sdio:c*v02D0dA887*
depends:        brcmutil,cfg80211
name:           brcmfmac
vermagic:       4.19.66-Re4son-v7+ SMP mod_unload modversions ARMv7 p2v8 
parm:           txglomsz:Maximum tx packet chain size [SDIO] (int)
parm:           debug:Level of debug output (int)
parm:           p2pon:Enable legacy p2p management functionality (int)
parm:           feature_disable:Disable features (int)
parm:           alternative_fw_path:Alternative firmware path (string)
parm:           fcmode:Mode of firmware signalled flow control (int)
parm:           roamoff:Do not use internal roaming engine (int)
parm:           iapp:Enable partial support for the obsoleted Inter-Access Point Protocol (int)
parm:           ignore_probe_fail:always succeed probe for debugging (int)

DrSchottky, 2019-11-11:
Yes, you have to use `4.19.66-Re4son-v7l+` headers. Don't you have that folder under /usr/src/?

andrewbeard, 2019-11-11:
Nope

DrSchottky, 2019-11-11:
Oh, that's he problem.
Is `kalipi-kernel-headers` package installed? if so, i think you need to download re4son kernel source and make the header pkg for your arch.
@evilsocket are there any other reasons beyond built-in nexmon for using kalipi-kernel?

andrewbeard, 2019-11-11:
Yup, the package is installed. For some reason it sticks everything in linux-headers-4.19.66-Re4son-v7+ and not linux-headers-4.19.66-Re4son-v7l+, though. It looks like that might just be an endianess flag, so I'd be a little surprised if the source was different. I don't think I have an actual kernel config to tell it otherwise, though.

andrewbeard, 2019-11-11:
Just to add another complication is looks like was just a new kalipi kernel update pushed to the apt repos. Info looks the same, so I would be shocked if the problem didn't continue:

abeard@pwnpi4:/usr/src/nexmon $ uname -a
Linux pwnpi4 4.19.81-Re4son-v7l+ #1 SMP Wed Nov 6 13:10:35 AEDT 2019 armv7l GNU/Linux
abeard@pwnpi4:/usr/src/nexmon $ dmesg | grep 7.45
[    2.736258] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/6 wl0: Oct  8 2018 09:18:41 version 7.45.154 (nexmon.org: 2.2.2-195-g01e0-1) FWID 01-4fbe0b04

DrSchottky, 2019-11-11:
What happens if you try ignore vermagic by adding `-f` to insmod?

andrewbeard, 2019-11-11:
No luck:

root@pwnpi4:/usr/src/nexmon/patches/bcm43455c0/7_45_189/nexmon# insmod -f brcmfmac_4.19.y-nexmon/brcmfmac.ko
insmod: ERROR: could not insert module brcmfmac_4.19.y-nexmon/brcmfmac.ko: Invalid module format

andrewbeard, 2019-11-11:
So from what I can tell the source code for the kernels should be identical. I need the kernel config for the v7l+ version, though. None of the config extraction options seem to be built in, so I'm kind of at a loss as to where I can get it from.

andrewbeard, 2019-11-11:
It seems like this may be an issue specific to the pi4. From what I can tell it's the only system that uses the v7l+ kernel. The build issue is probably the same thing that happened here:
https://github.com/seemoo-lab/nexmon/issues/321

I'm kind of dead in the water at the moment without being able to find the right kernel source or config. It looks like the kernel is being installed from an apt repo here:
http://http.re4son-kernel.com/re4son/

But the re4son website is pretty out of date and so I'm failing miserably trying to find the source package that corresponds to the installed kernel so far. It doesn't not appear that there's a package for the kalipi full kernel source, just the headers (that I already have, and don't include v7l+).

DrSchottky, 2019-11-12:
Yes, v7l+ (where l stands for lpae) targets Rpi4 beacuse of its larger RAM. The problem is that the kalipi-kernel-headers package is incomplete since it contains only headers for v7+.
If you look at re4son GH repo you can find their kernel and build script, so you should (never tried) to build a preper headers deb for your arch.
But imho switching to a 3rd party kernel is more a hassle than a benefit, that's why I asked Simone if it's needed.

DrSchottky, 2019-11-12:
@andrewbeard I patched the buildscripts to include v7l+ headers in the package.
Try [this](https://mega.nz/#!QBdFXagR!Ezo7oiaKUbnRCZwngXs8s0AcTJQITVALKR0_2UIXmhM)
It's not tested (I don't have a kalipi setup available atm) and the builder has many bugs, so it's an attempt.

andrewbeard, 2019-11-12:
COLLECTING STATISTICS read /usr/src/nexmon/STATISTICS.md for more information
  COMPILING src/version.c => obj/version.o (details: log/compiler.log)
  GENERATING LINKER FILE gen/nexmon.pre => gen/nexmon.ld
  GENERATING LINKER FILE gen/nexmon.pre => gen/flashpatches.ld
  LINKING OBJECTS => gen/patch.elf (details: log/linker.log, log/linker.err)
  GENERATING MAKE FILE gen/nexmon.pre => gen/nexmon.mk
  GENERATING MAKE FILE gen/nexmon.pre => gen/flashpatches.mk
  APPLYING FLASHPATCHES gen/flashpatches.mk => brcmfmac43455-sdio.bin (details: log/flashpatches.log)
  APPLYING PATCHES gen/nexmon.mk => brcmfmac43455-sdio.bin (details: log/patches.log)
  BUILDING DRIVER for kernel 4.19 brcmfmac_4.19.y-nexmon/brcmfmac.ko (details: log/driver.log)
/bin/sh: 1: ./scripts/recordmcount: not found
make[2]: *** [scripts/Makefile.build:303: /usr/src/nexmon/patches/bcm43455c0/7_45_189/nexmon/brcmfmac_4.19.y-nexmon/chip.o] Error 127
make[2]: *** Deleting file '/usr/src/nexmon/patches/bcm43455c0/7_45_189/nexmon/brcmfmac_4.19.y-nexmon/chip.o'
make[2]: *** Waiting for unfinished jobs....
/bin/sh: 1: ./scripts/recordmcount: not found
make[2]: *** [scripts/Makefile.build:303: /usr/src/nexmon/patches/bcm43455c0/7_45_189/nexmon/brcmfmac_4.19.y-nexmon/cfg80211.o] Error 127
make[2]: *** Deleting file '/usr/src/nexmon/patches/bcm43455c0/7_45_189/nexmon/brcmfmac_4.19.y-nexmon/cfg80211.o'
make[1]: *** [Makefile:1519: _module_/usr/src/nexmon/patches/bcm43455c0/7_45_189/nexmon/brcmfmac_4.19.y-nexmon] Error 2
make: *** [Makefile:52: brcmfmac.ko] Error 2

root@pwnpi4:/usr/src/nexmon/patches/bcm43455c0/7_45_189/nexmon# file /usr/src/linux-headers-4.19.81-Re4son-v7l+/scripts/recordmcount
/usr/src/linux-headers-4.19.81-Re4son-v7l+/scripts/recordmcount: ELF 64-bit LSB shared object, x86-64, version 1 (SYSV), dynamically linked, interpreter /lib64/ld-linux-x86-64.so.2, BuildID[sha1]=6312ca456fa4159f107061596ddecef734f07a9b, for GNU/Linux 3.2.0, not stripped

It looks like the scripts folder in the headers has locally compiled binaries, but since the headers were cross-compiled on an x64 system it's causing the nexmon build to shit the bed.

hackabean, 2019-11-12:
Hey @DrSchottky I would like to help if I can as well. 

**Hardware:**
Raspberry Pi 0w

**Output from uname -a:**
Linux pwnagotchi 4.19.66-Re4son+ #1 Sun Aug 18 13:42:02 AEST 2019 armv6l GNU/Linux

**dmesg | grep 7.45:**
[   14.339006] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM43430/1 wl0: Aug  7 2017 00:46:29 version 7.45.41.46 (r666254 CY) FWID 01-f8a78378

**modinfo /lib/modules/4.19.66-Re4son+/kernel/drivers/net/wireless/broadcom/brcm80211/brcmfmac/brcmfmac.ko:**

filename:       /lib/modules/4.19.66-Re4son+/kernel/drivers/net/wireless/broadcom/brcm80211/brcmfmac/brcmfmac.ko
license:        Dual BSD/GPL
description:    Broadcom 802.11 wireless LAN fullmac driver.
author:         Broadcom Corporation
firmware:       brcm/brcmfmac4373-sdio.bin
firmware:       brcm/brcmfmac4356-sdio.bin
firmware:       brcm/brcmfmac4354-sdio.bin
firmware:       brcm/brcmfmac43455-sdio.bin
firmware:       brcm/brcmfmac43430-sdio.bin
firmware:       brcm/brcmfmac43430a0-sdio.bin
firmware:       brcm/brcmfmac4339-sdio.bin
firmware:       brcm/brcmfmac43362-sdio.bin
firmware:       brcm/brcmfmac4335-sdio.bin
firmware:       brcm/brcmfmac43341-sdio.bin
firmware:       brcm/brcmfmac43340-sdio.bin
firmware:       brcm/brcmfmac4334-sdio.bin
firmware:       brcm/brcmfmac4330-sdio.bin
firmware:       brcm/brcmfmac4329-sdio.bin
firmware:       brcm/brcmfmac43241b5-sdio.bin
firmware:       brcm/brcmfmac43241b4-sdio.bin
firmware:       brcm/brcmfmac43241b0-sdio.bin
firmware:       brcm/brcmfmac43143-sdio.bin
firmware:       brcm/brcmfmac4373.bin
firmware:       brcm/brcmfmac43569.bin
firmware:       brcm/brcmfmac43242a.bin
firmware:       brcm/brcmfmac43236b.bin
firmware:       brcm/brcmfmac43143.bin
srcversion:     389E0869FFB2AAD06E0D34E
alias:          usb:v04B4p0BDCd*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0A5Cp0BDCd*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v043Ep3101d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v04B4pBD29d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v13B1p0039d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0A5CpBD27d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0A5CpBD1Fd*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0A5CpBD17d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0A5CpBD1Ed*dc*dsc*dp*ic*isc*ip*in*
alias:          sdio:c*v02D0d4373*
alias:          sdio:c*v02D0d4356*
alias:          sdio:c*v02D0d4354*
alias:          sdio:c*v02D0dA9BF*
alias:          sdio:c*v02D0d4345*
alias:          sdio:c*v02D0dA9A6*
alias:          sdio:c*v02D0d4339*
alias:          sdio:c*v02D0d4335*
alias:          sdio:c*v02D0dA9A4*
alias:          sdio:c*v02D0dA962*
alias:          sdio:c*v02D0dA94D*
alias:          sdio:c*v02D0dA94C*
alias:          sdio:c*v02D0d4334*
alias:          sdio:c*v02D0d4330*
alias:          sdio:c*v02D0d4329*
alias:          sdio:c*v02D0d4324*
alias:          sdio:c*v02D0dA887*
depends:        brcmutil,cfg80211
intree:         Y
name:           brcmfmac
vermagic:       4.19.66-Re4son+ mod_unload modversions ARMv6 p2v8 
parm:           txglomsz:Maximum tx packet chain size [SDIO] (int)
parm:           debug:Level of debug output (int)
parm:           p2pon:Enable legacy p2p management functionality (int)
parm:           feature_disable:Disable features (int)
parm:           alternative_fw_path:Alternative firmware path (string)
parm:           fcmode:Mode of firmware signalled flow control (int)
parm:           roamoff:Do not use internal roaming engine (int)
parm:           iapp:Enable partial support for the obsoleted Inter-Access Point Protocol (int)
parm:           ignore_probe_fail:always succeed probe for debugging (int)

andrewbeard, 2019-11-12:
I figured I'd give it one more jackass try and replace all the x64 script binaries with ARM ones from my local build tree. They shouldn't be dependent on the patchset, right?

Module built (yay!), but still wouldn't insert (boo):

  COLLECTING STATISTICS read /usr/src/nexmon/STATISTICS.md for more information
  COMPILING src/version.c => obj/version.o (details: log/compiler.log)
  GENERATING LINKER FILE gen/nexmon.pre => gen/nexmon.ld
  GENERATING LINKER FILE gen/nexmon.pre => gen/flashpatches.ld
  LINKING OBJECTS => gen/patch.elf (details: log/linker.log, log/linker.err)
  GENERATING MAKE FILE gen/nexmon.pre => gen/nexmon.mk
  GENERATING MAKE FILE gen/nexmon.pre => gen/flashpatches.mk
  APPLYING FLASHPATCHES gen/flashpatches.mk => brcmfmac43455-sdio.bin (details: log/flashpatches.log)
  APPLYING PATCHES gen/nexmon.mk => brcmfmac43455-sdio.bin (details: log/patches.log)
  BUILDING DRIVER for kernel 4.19 brcmfmac_4.19.y-nexmon/brcmfmac.ko (details: log/driver.log)
  COPYING brcmfmac43455-sdio.bin => /lib/firmware/brcm/brcmfmac43455-sdio.bin
  RELOADING brcmfmac
insmod: ERROR: could not insert module brcmfmac_4.19.y-nexmon/brcmfmac.ko: Invalid module format
make: *** [Makefile:190: install-firmware] Error 1
root@pwnpi4:/usr/src/nexmon/patches/bcm43455c0/7_45_189/nexmon# insmod -f brcmfmac_4.19.y-nexmon/brcmfmac.ko 
insmod: ERROR: could not insert module brcmfmac_4.19.y-nexmon/brcmfmac.ko: Invalid module format

DrSchottky, 2019-11-12:
This header thing is becoming a pain in the ass.
Since pwnagotchi [can be installed on a working system](https://pwnagotchi.ai/installation/#installing-on-any-gnu-linux) why don't we burn a clean Raspbian Lite image, do a couple of nexmon tests and then install the software?

andrewbeard, 2019-11-12:
Something I just noticed, that failedinstall  process looks like it _did_ overwrite the firmware in /lib/firmware/brcm/brcmfmac43455-sdio.bin and when I rebooted the pi it did load the new firmware. 

```
[    2.664068] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[    2.673602] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/6 wl0: Nov 12 2019 09:40:03 version 7.45.189 (nexmon.org: 2.2.2-273-gaaa80e-dirty-7) FWID 01-e1db26e2
```

Unfortunately the channel change errors now start _immediately_, about 14 second after I boot:

```
[   14.299243] brcmfmac: brcmf_cfg80211_nexmon_set_channel: Set Channel failed: chspec=53286, -52
[   15.312016] brcmfmac: brcmf_cfg80211_nexmon_set_channel: Set Channel failed: chspec=53290, -52
[   16.325362] brcmfmac: brcmf_cfg80211_nexmon_set_channel: Set Channel failed: chspec=53294, -52
[   30.549909] brcmfmac: brcmf_cfg80211_nexmon_set_channel: Set Channel failed: chspec=53286, -52
[   31.567560] brcmfmac: brcmf_cfg80211_nexmon_set_channel: Set Channel failed: chspec=53290, -52
[   32.584209] brcmfmac: brcmf_cfg80211_nexmon_set_channel: Set Channel failed: chspec=53294, -52
[   62.261185] brcmfmac: brcmf_cfg80211_nexmon_set_channel: Set Channel failed: chspec=53286, -52
[   63.278705] brcmfmac: brcmf_cfg80211_nexmon_set_channel: Set Channel failed: chspec=53290, -52
[   64.299974] brcmfmac: brcmf_cfg80211_nexmon_set_channel: Set Channel failed: chspec=53294, -52
[   78.550006] brcmfmac: brcmf_cfg80211_nexmon_set_channel: Set Channel failed: chspec=53286, -52
[   79.568249] brcmfmac: brcmf_cfg80211_nexmon_set_channel: Set Channel failed: chspec=53290, -52
[   80.586455] brcmfmac: brcmf_cfg80211_nexmon_set_channel: Set Channel failed: chspec=53294, -52
[  128.297310] brcmfmac: brcmf_cfg80211_nexmon_set_channel: Set Channel failed: chspec=53286, -52
[  131.399334] brcmfmac: brcmf_cfg80211_nexmon_set_channel: Set Channel failed: chspec=53286, -52
[  134.503219] brcmfmac: brcmf_cfg80211_nexmon_set_channel: Set Channel failed: chspec=53286, -52
[  137.631244] brcmfmac: brcmf_cfg80211_nexmon_set_channel: Set Channel failed: chspec=53286, -52
[  140.753593] brcmfmac: brcmf_cfg80211_nexmon_set_channel: Set Channel failed: chspec=53286, -52
```

It looks like new firmware + old module is even worse than old firmware + old module.

andrewbeard, 2019-11-12:
@DrSchottky  Sounds like a good plan. I'll see if I've got a spare sd card I can start from.

andrewbeard, 2019-11-12:
OK, I've got a fresh Raspbian lite image + the nexmon driver installed with the 7.45.189 firmware. What do you need me to try?

DrSchottky, 2019-11-12:
install aircrack-ng, remove wpasupplicant, activate mon0 and let airodump run for a while.

andrewbeard, 2019-11-13:
So, I tried a couple things last night:
- airodump seems to run without error for 6+ hours, but it picks up very little. Few stations detected. 
- I tried running bettercap but it produces an error when initializing the interface complaining that it couldn't set the channel
- ```
root@pwnpi4-ng:/home/abeard# iwconfig mon0 channel 1
Error for wireless request "Set Frequency" (8B04) :
    SET failed on device mon0 ; Device or resource busy.
```

So, the good news is that it isn't locking up. The bad news is that I'm not really sure I'm getting to the  fully functional point (even without trying injection).

DrSchottky, 2019-11-13:
Ok, airodump can't find many stations beacuse it can't do channel hopping.
So reboot your device and make sure qpa_supplicant isn't running/isn't installed and start mon0 this way
```
sudo ip link set dev wlan0 down
sudo iwconfig wlan0 channel 6
sudo iw phy `iw dev wlan0 info | gawk '/wiphy/ {printf "phy" $2}'` interface add mon0 type monitor
sudo ip link set dev mon0 up
sudo iwconfig mon0 channel 6
```
Now you should have mon0 set on channel 6 (and you should be able to change the channel with iwconfig), so start airdoump telling it to work on both 2.4 and 5.8 bands
`sudo airodump-ng --band abg mon0`

andrewbeard, 2019-11-13:
MUCH better. It immediately popped up with 5x more networks than it
detected all night. I'll let this run for a bit and see if it craps out.

[38 more replies]