evilsocket/pwnagotchi #270: Return value from plugin.on()

cdiemel, 2019-10-12:
I am attempting to develop a plugin, however I cannot figure out a good way to get data from the plugin back to the main loop of code.  

I would like to add a few lines of code that look to see if the plugin function that was called returned a value, if it did then pass that value back to the function that called `plugin.on()`

in pwnagotchi/pwnagotchi/plugins/__init__.py  - lines 14-23
```
def on(event_name, *args, **kwargs):
    global loaded
    cb_name = 'on_%s' % event_name
    for plugin_name, plugin in loaded.items():
        if cb_name in plugin.__dict__:
            # print("calling %s %s(%s)" %(cb_name, args, kwargs))
            try:
                plugin.__dict__[cb_name](*args, **kwargs)
            except Exception as e:
logging.error("error while running %s.%s : %s" % (plugin_name, cb_name, e))
```
I would like to change to the following:
```
def on(event_name, *args, **kwargs):
    global loaded
    cb_name = 'on_%s' % event_name
    for plugin_name, plugin in loaded.items():
        if cb_name in plugin.__dict__:
            # print("calling %s %s(%s)" %(cb_name, args, kwargs))
            try:
                ret_val = plugin.__dict__[cb_name](*args, **kwargs)
                if ret_val is not None:
                    return ret_val
            except Exception as e:
logging.error("error while running %s.%s : %s" % (plugin_name, cb_name, e))
```
I have not had a chance to test this code, however I was outlining the code for a plugin and noticed, with the help of @dadav, that I could not get a value back. 

I have not discovered any other alternatives to try, but am open to learning more/better ways to program.

evilsocket, 2019-10-12:
why would you want to do that exactly? what's the usecase?

cdiemel, 2019-10-12:
I want to save a last upload time to a file using the utils.status function.  However I need to check the value of the last upload against the current time every time there is a list of APs polled in agent.py
```
    def get_access_points(self):
        whitelist = self._config['main']['whitelist']
        aps = []
        try:
            s = self.session()
            for ap in s['wifi']['aps']:
## --- I want to check the time here --- ##
                if ap['hostname'] not in whitelist:
                    if self._filter_included(ap):
                        aps.append(ap)
        except Exception as e:
            logging.exception("error")

        aps.sort(key=lambda ap: ap['channel'])
        return self.set_access_points(aps)
```

However I believe doing a file read call every time there is new list of access points is time and resource intensive.  I would like to init the plugin and read in the last time there was an upload and then keep that stored in a local variable.  

however while writing this i just saw that there is a function I didn't see last night, that is called in the plugin when there is a new list of APs.

It seems I can look through the list of provided APs and find the specific one im looking for and act on them accordingly.

evilsocket, 2019-10-12:
can you explain more at the higher level what your plugin would be doing?

cdiemel, 2019-10-12:
I saw this feature request
https://github.com/evilsocket/pwnagotchi/issues/207
and it was something that I had been thinking about doing myself and decided to try to write it as a plugin for the community.  

The idea is to go for a long walk with your pwnagotchi and when you return and are close enough to the specified network, it stops collecting handshakes and transitions into managed mode allowing you to connect to your network and upload config files, and other data.  

I was going to include the option to set a max time that the device is in managed mode.  The original author wanted this to be able to SSH into the device when on their home network, I would like to take it a step further and allow automated upload of config files, etc, much like the auto-backup plugin.  

However with this being similar to the auto-backup plugin I was considering the idea of just using it as a way to provide internet, having the plugin control all of the ifup/down of interfaces and DHCP configuration with the specified network.

0xRoM, 2019-10-12:
@cdiemel can recommend this guide for intro on how to do a pull request: https://opensource.com/article/19/7/create-pull-request-github

cdiemel, 2019-10-13:
In retrospect, after reading through more of the code and how the plugin system works, I believe a different solution would work better and provide additional functionality for the project as a whole.

By adding an additional plugin function `plugin.on("unfiltered_ap_list")` it would my plugin and others the ability to take actions based on the raw access point list.  This also negates the need to have a return value from the `plugin.on()` function.  

```
    def get_access_points(self):
        whitelist = self._config['main']['whitelist']
        aps = []
        try:
            s = self.session()
## --
            plugins.on("unfiltered_wifi_list", s['wifi']['aps')
## --
            for ap in s['wifi']['aps']:
                if ap['hostname'] not in whitelist:
                    if self._filter_included(ap):
                        aps.append(ap)
        except Exception as e:
            logging.exception("error")

        aps.sort(key=lambda ap: ap['channel'])
        return self.set_access_points(aps)
```

I'm not sure if I should open a  different issue for this change of course or if I should still submit a PR under this issue.

closed 2019-10-14 by evilsocket