evilsocket/pwnagotchi #273: backup.sh throws permission error on unmodified images

incredincomp, 2019-10-13:
**Is your feature request related to a problem? Please describe.**
When running sudo ./backup.sh on a new image, the backup.sh script calls scp directed at root. This will result in a permission error due to root having sshlogin turned off by default on all new raspbian releases.

**Describe the solution you'd like**
Should probably just scp at user pi so we can have permission to log in with scp.

**Describe alternatives you've considered**
Adding info into the docs about installing your host machines ssh key and running sshd

**Additional context**
I ended up just changing my script to pi@ and it works fine last time i tried, that was rc2 tho.

evilsocket, 2019-10-13:
simply don't run it with sudo

closed 2019-10-13 by evilsocket

incredincomp, 2019-10-14:
```bash
incredincomp@some-laptop[~/git_stuff/theirs/pwnagotchi/scripts]:$ ./backup.sh 10.0.0.2 ~/pwn_bak/cr33p3r/
@ backing up 10.0.0.2 to /home/incredincomp/pwn_bak/cr33p3r/ ...
  /root/brain.nn -> /tmp/pwnagotchi_backup/root/
root@10.0.0.2's password: 
Permission denied, please try again.
root@10.0.0.2's password: 
```
I updated the docs on the site to specifically say this script will not work without ssh_keys being present on the pi. My initial suggestion wont even work anyway. We have to make people install ssh keys somehow.. more intensely maybe

evilsocket, 2019-10-14:
It works if you use the right password dude, it’s just ash.

incredincomp, 2019-10-14:
damnit im logged into it already. Okay, well duh. sorry for wasting your time. 

since its just ssh tho, can we do backup.sh over 2222 or something?