evilsocket/pwnagotchi #284: [BUG] Pcap files under /root/handshakes/ are saved that contain no handshakes

wy4h5evs4eve4s, 2019-10-13:
**Describe the bug**
After running pwnagotchi for a day I noticed that nearly half of files saved under `/root/handshakes/` contain no handshakes in them. There are packets in them, but only about 23 out of 70 files contain hanshakes among these packets.

Example:
```
$ aircrack-ng foobar.pcap
Opening foobar.pcap
Read 12 packets.

   #  BSSID              ESSID                     Encryption

   1  XX:XX:XX:XX:XX:X  foobar           WPA (0 handshake)

Choosing first network as target.

Opening foobar.pcap
Read 12 packets.

1 potential targets

Please specify a dictionary (option -w).
```

Is that intentional? Are packet dumps supposed to be written there even though no handshakes have been caught?

**Expected behavior**
I'd expect no `.pcap` file to be saved under `/root/handshakes` for a particular AP if no handshakes have been obtained for the AP. I'd expect every file in that folder to contain handshakes.

**Environment (please complete the following information):**
 - Pwnagotchi version: v1.0.0RC3
 - OS version: Pwnagotchi v1.0.0RC3 image from Github
 - Type of hardware: RPi0W

**Additional context**
This behavior is causing additional consequences for the `onlinehashcrack` plugin. A file containing 0 hanshakes will still get marked as uploaded.

dadav, 2019-10-14:
You could use the AircrackOnly plugin...

evilsocket, 2019-10-14:
The fact that aircrack can’t find handshakes doesn’t mean there aren’t any. Aircrack is not the only software.

closed 2019-10-14 by evilsocket

maichai, 2021-03-30:
The problem is not in aircrack-ng. Aircrack-ng actually properly identified the handshakes. However, aircrackonly is not called for each and every pcap that is captured, so it cannot sort out the "bad" pcaps. The is a problem with the call mechanism. you will see that for the "bad" pcaps there is no log message from aircrackonly in the log, but it does always log something for every pcap it is processing.