evilsocket/pwnagotchi #309: better time sync system

xenDE, 2019-10-16:
it can happen that a pwnagotchi is only restarted after several days and gets internet synchronization after several hours.
in these hours all stored timestamps are wrong.

for this an idea like these wrong timestamps, e.g. with stored net-pos data afterwards could be corrected:

boot-uuid: 1d234f45				<< uuid created on every boot
	boot-uptime: 13.31s			<< uptime at save boot-ts
		boot-ts: 1571227347		<< system time
	sync-uptime: 1839.23s		<< uptime on timesync/ntp sync or NULL
		synced-ts: 1571227538	<< new synced timestamp

- save boot-uuid + boot-uptime + boot-ts values as /var/times-syncs/[boot-uuid].json and boot-uuid as /tmp/boot-uuid on boot
- create a hook for first time sync (must not miss any time sync) and set sync-uptime+synced-ts in /var/times-syncs/[boot-uuid].json and "touch /tmp/time-synced"
- on save wlan-position files always save uptime+timestamp+isSynced
- so we can calculate fixed timestamps
- better/accurat uptime on screen
- idea: disable nptd service by default and use a timesync plugin for first sync & start ntpd on_internet_available()

what do you think?
Did I miss something?

xenDE, 2019-10-16:
> i'm a bit worried about the "must not miss any time sync" part

because i dont know in what ways the raspbian is syncing the time.

ntp, gps-time, ???

but after thinking about:
no hook needed, if the time was changed/synced and changed with e.g. over 1 minute offset, it was synced and can be set to "synced"

i will try to code this on weekend.

dadav, 2019-10-17:
is related to #308

dadav, 2019-10-17:
As far as i know `systemd-timesyncd` is doing the synchronisation.
Also, `systemd-timesyncd` touches `/var/lib/systemd/clock`, when the sync happened.

xenDE, 2019-10-17:
nice, and the boot-uuid are there too: md5(/usr/lib/systemd/random-seed)

xenDE, 2019-10-19:
first test looks good, time offset is correct calculated after time synced with over 60seconds offset.
but needs more time to be finished+integrated - i think not as plugin, it needs to be in core.

https://github.com/xenDE/pwnagotchi-plugin-timesync

everybody welcome: https://github.com/xenDE/pwnagotchi-plugin-timesync

xenDE, 2019-11-03:
i added a test, and it looks working.

https://github.com/xenDE/pwnagotchi-plugin-timesync

for test:
 - run on-boot.sh one time or insert in /etc/rc.local
 - run test.py

xenDE, 2019-11-20:
an RTC is a cheap and easy solution:
- https://pwnagotchi.ai/installation/#hardware-clock
- https://community.pwnagotchi.ai/search?q=rtc%20order%3Alatest

closed 2019-11-20 by xenDE