evilsocket/pwnagotchi #311: Evil Twin AP

myasn1k, 2019-10-16:
It would be awesome to implement a plugin to perform automatic [evil twin attack](https://en.wikipedia.org/wiki/Evil_twin_(wireless_networks))!
Surely another wireless interface would be required, but with a cheap wifi adapter the problem should be solved.  
Maybe the plugin could implement a timer that checks if connected to the data port there is a wireless interface (or maybe it could be done with interrupts?).
The only drawbacks I found about this solution is that the attack is based in part on social engineering, so it could not work.

Thanks for the attention and thanks a lot to the dev-team for this beautiful project!

PeterAlfonsLoch, 2019-10-27:
I advise against it, to be put in this repo. the mere exisance of that code on a device is illegal in many countries. Now If you or someone want's to wite it as a plugin then it is upon the one installing it to check if it is legal or not. The point of wanting to show how insecure something is, is very different from actually wanting to steal passwords. Even from an ethical viewpoint i'd advice against it to be put in this repo aka in the RC

myasn1k, 2019-10-27:
Thanks a lot for the reply!I completely understand what you're saying and you're right; I thought it would have been useful in context like Red teaming (maybe? I'm not an expert) but as you said it could be weaponized easily and used for purposes that may not be so legal or ethical..So sorry to not have thought about this in advance!-------- Messaggio originale --------Oggetto: Re: [evilsocket/pwnagotchi] Evil Twin AP (#311)Da: PeterAlfonsLoch A: evilsocket/pwnagotchi CC: Pietro Mazzini ,Author I advise against it, to be put in this repo. the mere exisance of that code on a device is illegal in many countries. Now If you or someone want's to wite it as a plugin then it is upon the one installing it to check if it is legal or not. The point of wanting to show how insecure something is, is very different from actually wanting to steal passwords. Even from an ethical viewpoint i'd advice against it to be put in this repo aka in the RC

[
{
"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/evilsocket/pwnagotchi/issues/311?email_source=notifications\u0026email_token=AIWL3HPGFWA26JCJ773MYQTQQVU4XA5CNFSM4JBOP4GKYY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOECK4GUQ#issuecomment-546685778",
"url": "https://github.com/evilsocket/pwnagotchi/issues/311?email_source=notifications\u0026email_token=AIWL3HPGFWA26JCJ773MYQTQQVU4XA5CNFSM4JBOP4GKYY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOECK4GUQ#issuecomment-546685778",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]

PeterAlfonsLoch, 2019-10-27:
I am just one of many users i am not party of the deciding force but if someone would create a seperate git for a plugin of that kind then yes why not. I just wanted to give my 2 cents about it. i think that the project itself should remain as legal as possible. For now it is, depending on the country, mor of a gray area.

But your idea could be possible.

gompa, 2019-12-18:
isn't the whole deauth part "illegal" ? 
considering : https://www.fcc.gov/document/warning-wi-fi-blocking-prohibited
so at least in the usa,but there are probably more countries with similar laws

myasn1k, 2019-12-18:
Little update.
I thought about the whole idea; implementing an evil twin attack that automatically redirects new connected users to a phishing webpage to steal wifi credentials is a bit "crude".
Not just by the legal point of view that, as @gompa points to, maybe isn't so relevant; but in the sense that every AP is different from the others, so the attacker would have to craft a web page ad-hock for every AP (and that's not so smart because the automatic part of the plugin scope in this way just blows up). 
So the solutions are: 
- Best solution: make a plugin that fully implements evil twin and write also a script to find infos about the attacked AP just by knowing the SSID (something like googling infos and images connected with the SSID)
    - Comment: not so cool, won't work from my point of view
- Something like: create a plugin that implements [EAPHammer](https://github.com/s0lst1c3/eaphammer).
    - Comment: yes, this isn't the best solution (it'll just work with enterprise wifi networks), but at least is fully automatable (i hope..).
Tell me what you think, now I'm studying for university exams so I won't work on this til maybe march/april.
I hope that this discussion will make someone think about this opportunity and work about it :)

bruvv, 2019-12-28:
https://github.com/evilsocket/pwnagotchi/pull/757 it's here ;)

myasn1k, 2019-12-28:
#757 exactly what I was searching for! :)

closed 2019-12-28 by myasn1k