evilsocket/pwnagotchi #320: Add themes functionality

dadav, 2019-10-18:
The idea is to pack informations like widgets, voice and faces into "themes".

There could be a config like:
```yaml
main:
  theme: default # face-only, minimal, etc.
```

What do you think of this?

dipsylala, 2019-10-18:
Love it :D Could this be done in stages? 
1) Faces
2) Voice
3) Layout

Reason being - 3 is highly dependant on displays and might need a bigger discussion - a theme might need multiple layouts if it were to be used with multiple displays. We could perhaps go for a similar approach used in RetroPie/Kodi where we have layouts optimised for particular displays? eg Minimal_InkyPhat, Minimal_WaveShare2?

evilsocket, 2019-10-19:
from the next releases, yes, the entire UI will be customizable, faces too ;)

evilsocket, 2019-10-20:
it's a start https://github.com/evilsocket/pwnagotchi/commit/cd5d783c52cedaab09d8f5ad1bbb7b62eec91401

incredincomp, 2019-10-20:
This is awesome and thank you so much @evilsocket

kbeflo, 2019-11-09:
Please allow changing font and background colors for the web interface too