evilsocket/pwnagotchi #322: [BUG] /root/pwnagotchi.png not updated

michelep, 2019-10-18:
**Describe the bug**
seems that /root/pwnagotchi.png file were not updated.

**To Reproduce**
vanilla installation of pwnagotchi?

**Expected behavior**
/root/pwnagotchi.png should update

**Screenshots**
https://twitter.com/michele_pinassi/status/1185104264659648512

**Environment (please complete the following information):**
 - Pwnagotchi version: latest
 - OS version: raspbian
 - Type of hardware: RPIw0
 - Any additional hardware used: LCD 3,5" Waveshare display

evilsocket, 2019-10-18:
the file is updated at every UI refresh as you can see here https://github.com/evilsocket/pwnagotchi/blob/master/pwnagotchi/ui/display.py#L56

in the tweet i see your unit in manual mode, in manual mode the UI doesn't change, so why the file should be updated? am i missing something?

michelep, 2019-10-18:
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/dist-packages/pwnagotchi/ui/display.py", line 225, in _inky_render
    self._display.show()
  File "/usr/local/lib/python3.7/dist-packages/inky/inky.py", line 325, in show
    self._update(buf_a, buf_b, busy_wait=busy_wait)
  File "/usr/local/lib/python3.7/dist-packages/inky/inky.py", line 238, in _update
    self.setup()
  File "/usr/local/lib/python3.7/dist-packages/inky/inky.py", line 213, in setup
    self._spi.open(0, self.cs_pin)
FileNotFoundError: [Errno 2] No such file or directory

evilsocket, 2019-10-18:
i think that's a separate issue that's been fixed by a PR i merged a few minutes ago

michelep, 2019-10-18:
TNX!

closed 2019-10-18 by evilsocket