evilsocket/pwnagotchi #323: [BUG]

michelep, 2019-10-18:
**Describe the bug**
on syslog i got:

> Oct 18 14:19:08 localhost pwnagotchi-launcher[601]: [2019-10-18 14:19:08,558] [ERROR] error while rendering on inky
> Oct 18 14:19:08 localhost pwnagotchi-launcher[601]: Traceback (most recent call last):
> Oct 18 14:19:08 localhost pwnagotchi-launcher[601]:   File "/usr/local/lib/python3.7/dist-packages/pwnagotchi/ui/hw/inky.py", line 67, in render
> Oct 18 14:19:08 localhost pwnagotchi-launcher[601]:     self._display.show()
> Oct 18 14:19:08 localhost pwnagotchi-launcher[601]:   File "/usr/local/lib/python3.7/dist-packages/inky/inky.py", line 325, in show
> Oct 18 14:19:08 localhost pwnagotchi-launcher[601]:     self._update(buf_a, buf_b, busy_wait=busy_wait)
> Oct 18 14:19:08 localhost pwnagotchi-launcher[601]:   File "/usr/local/lib/python3.7/dist-packages/inky/inky.py", line 238, in _update
> Oct 18 14:19:08 localhost pwnagotchi-launcher[601]:     self.setup()
> Oct 18 14:19:08 localhost pwnagotchi-launcher[601]:   File "/usr/local/lib/python3.7/dist-packages/inky/inky.py", line 213, in setup
> Oct 18 14:19:08 localhost pwnagotchi-launcher[601]:     self._spi.open(0, self.cs_pin)
> Oct 18 14:19:08 localhost pwnagotchi-launcher[601]: FileNotFoundError: [Errno 2] No such file or directory
> Oct 18 14:19:10 localhost kernel: [ 5632.888050] brcmfmac: brcmf_cfg80211_nexmon_set_channel: Set Channel failed: chspec=4103, -110
> Oct 18 14:19:13 localhost kernel: [ 5635.768143] brcmfmac: brcmf_cfg80211_nexmon_set_channel: Set Channel failed: chspec=4097, -110
> Oct 18 14:19:17 localhost kernel: [ 5639.688078] brcmfmac: brcmf_cfg80211_nexmon_set_channel: Set Channel failed: chspec=4103, -110
> Oct 18 14:19:22 localhost kernel: [ 5645.048153] brcmfmac: brcmf_cfg80211_nexmon_set_channel: Set Channel failed: chspec=4103, -110
> 

**Environment (please complete the following information):**
 - Pwnagotchi version: latest 1.0.0RC5
 - Type of hardware: RPIW0
 - Any additional hardware used: Waveshare 2.4inch LCD, Raspi UPS HAT

**Additional context**

This bug doesnt' seem to affect functionality.

evilsocket, 2019-10-18:
the crash seems related to the inky library not finding the screen ... is it correctly connected?

closed 2019-10-18 by evilsocket

oglok, 2020-03-22:
I found the same  error but with the 3.5" LCD Waveshare screen:

`22:08:29 [ERROR] error while rendering on inky
(most recent call last):
 File "/usr/local/lib/python3.7/dist-packages/pwnagotchi/ui/hw/inky.py", line 77, in render
   self._display.show()
 File "/usr/local/lib/python3.7/dist-packages/inky/inky.py", line 325, in show
   self._update(buf_a, buf_b, busy_wait=busy_wait)
 File "/usr/local/lib/python3.7/dist-packages/inky/inky.py", line 238, in _update
   self.setup()
 File "/usr/local/lib/python3.7/dist-packages/inky/inky.py", line 213, in setup
   self._spi.open(0, self.cs_pin)
[Errno 2] No such file or directory`

ctyzan, 2022-03-30:
the same error rpi4b 8gb