evilsocket/pwnagotchi #332: [BUG] FileNotFoundError with Papirus displays

evilsocket, 2019-10-19:
```pi@pwnagotchi:~ $ systemctl status pwnagotchi
● pwnagotchi.service - pwnagotchi Deep Reinforcement Learning instrumenting bettercap for WiFI pwning.
   Loaded: loaded (/etc/systemd/system/pwnagotchi.service; enabled; vendor preset: enabled)
   Active: activating (auto-restart) (Result: exit-code) since Wed 2019-07-10 01:23:32 BST; 3s ago
     Docs: https://pwnagotchi.ai
  Process: 634 ExecStart=/usr/bin/pwnagotchi-launcher (code=exited, status=1/FAILURE)
 Main PID: 634 (code=exited, status=1/FAILURE)
pi@pwnagotchi:~ $ journalctl -fu pwnagotchi
-- Logs begin at Wed 2019-07-10 01:21:16 BST. --
Jul 10 01:23:32 pwnagotchi pwnagotchi-launcher[634]:     self._implementation.initialize()
Jul 10 01:23:32 pwnagotchi pwnagotchi-launcher[634]:   File "/usr/local/lib/python3.7/dist-packages/pwnagotchi/ui/hw/papirus.py", line 39, in initialize
Jul 10 01:23:32 pwnagotchi pwnagotchi-launcher[634]:     self._display = EPD()
Jul 10 01:23:32 pwnagotchi pwnagotchi-launcher[634]:   File "/usr/local/lib/python3.7/dist-packages/pwnagotchi/ui/hw/libs/papirus/epd.py", line 65, in __init__
Jul 10 01:23:32 pwnagotchi pwnagotchi-launcher[634]:     self._lm75b = LM75B()
Jul 10 01:23:32 pwnagotchi pwnagotchi-launcher[634]:   File "/usr/local/lib/python3.7/dist-packages/pwnagotchi/ui/hw/libs/papirus/lm75b.py", line 23, in __init__
Jul 10 01:23:32 pwnagotchi pwnagotchi-launcher[634]:     self._bus = smbus.SMBus(busnum)
Jul 10 01:23:32 pwnagotchi pwnagotchi-launcher[634]: FileNotFoundError: [Errno 2] No such file or directory
Jul 10 01:23:32 pwnagotchi systemd[1]: pwnagotchi.service: Main process exited, code=exited, status=1/FAILURE
Jul 10 01:23:32 pwnagotchi systemd[1]: pwnagotchi.service: Failed with result 'exit-code'.
^C
pi@pwnagotchi:~ $```

caquino, 2019-10-19:
Derek assign: syshero

ukscone, 2019-10-20:
using raspi-config to enable spi and i2c seems to work although I think my display seems to be borked so can't check until i've checked that but everything seemed to start up as expected with no error. enabling in /boot/config.txt is probably the "recommended" fix unless it causes problems with other displays

ukscone, 2019-10-23:
found a working papirus screen in my pile and the enable i2c/enable spi lets the papirus display work