evilsocket/pwnagotchi #335: Clear legal section

pahakalle, 2019-10-20:
Readme should have a clear legal section about the possible illegality of dehauth/WiFi attacks.

incredincomp, 2019-10-20:
**IANAL**

The legal ramifications are entirely subjective to where the user lives and operates the pwnagotchi software. That being said, my understanding of the general international laws lead me to believe this software, on its own, with no user interference or modification, is completely legal. The cracking of passwords collected is also completely legal. 

The illegality comes in when you focus deauth attacks on a single device to render a DoS of sorts. 
Also, any access to any system that you do not own or do not have permission to access will be viewed as a crime. This would include logging into a WiFi network with a password you have cracked.

Currently, Pwnagotchi's will not send repeated deauths to the same client on the same network, avoiding the definition of DoS. IE. Legal

We also can't stop people from illegally accessing things they are not allowed to access, pwnagotchi does not do anything spectacular that you could not do with a laptop when trying to pwn a network on your own. If a user is cracking passwords to illegally access networks they are not allowed access to, it is the users responsibility to know their local laws, and receive permission before attempting access to networks that dont belong to them. The users actions define the legality here, not the pwnagotchi .img's or code library.

Just my two cents.
~incredincomp

evilsocket, 2019-10-26:
@incredincomp i couldn't have said it better