evilsocket/pwnagotchi #337: 'Includes' config section to act as SSID whitelist

dipsylala, 2019-10-20:
**Use Case**
The units would be great to use on site tests. However there may be contract/legal implications for deauthing and capturing handshakes that are considered out of scope, or we may wish for the unit to target specific SSIDs and save battery/computing power on anything else.

**Solution**
As a tester, I would like to be able to specifically include targets of interest - the ones in scope that I wish to audit.

** Alternatives**
In the past I've manually captured SSIDs then sent the specific ones for auditing to airreplay-ng, or done the configuration ahead of time on a laptop. Phone options may be available, but they lack the AI approach provided by pwnagotchi.

**Potential Solution**
Have an 'includes' section in the config in the same way we have an 'excludes'. If populated, the unit would consider this on first assessment of a SSID and dismiss it if it wasn't there. If handshakes for all the items in the 'Includes' section are captured, notify the user via 'Happy Face' and message, and stop searching.

If the includes section is missing or blank, the unit defaults to everything in scope unless it is in the excludes section - its current mode of operation.

dadav, 2019-10-20:
Isn't this not already possible with this? 
https://github.com/evilsocket/pwnagotchi/blob/2d7b6b54fe2401f5d2ad8ee8f65e44bf8d5efc23/pwnagotchi/defaults.yml#L90

dipsylala, 2019-10-20:
Oh crumbs, you’re right - I missed this in a later update. I’ll close this

closed 2019-10-20 by dipsylala