evilsocket/pwnagotchi #341: Backup script that doesn't require zip and scp

TheVladdestVlad, 2019-10-20:
I've ran into some issues using the original backup script when the pwnagotchi was dying (unable to connect to the internet any more) and the script didn't seem to recognize the root password to ssh into localhost.
I've modified the backup script so that it doesn't require a working network connection.
-Script
```
#!/usr/bin/env bash
# output backup tar file
OUTPUT=${1:-pwnagotchi-backup.tar.gz}
# temporary folder
TEMP_BACKUP_FOLDER=/tmp/pwnagotchi_backup
# what to backup
FILES_TO_BACKUP=(
  /root/brain.nn
    /root/brain.json
      /root/.api-report.json
        /root/handshakes
	  /etc/pwnagotchi/
	    /etc/hostname
	      /etc/hosts
	        /etc/motd
		  /var/log/pwnagotchi.log
		  )

		  echo "@ backing up $HOSTNAME to $OUTPUT ..."

		  #Delete previously existing backup dir
		  rm -rf "$TEMP_BACKUP_FOLDER"

		  for file in "${FILES_TO_BACKUP[@]}"; do
			    dir=$(dirname $file)
			      echo "  $file -> $TEMP_BACKUP_FOLDER$dir/"
			        mkdir -p "$TEMP_BACKUP_FOLDER/$dir"
				  cp -pr $file "$TEMP_BACKUP_FOLDER$dir/"
			  done

			  ZIPFILE=/home/pi/"$OUTPUT"
			  pushd $PWD
			  cd "$TEMP_BACKUP_FOLDER"
			  tar -cvzf "$ZIPFILE" . 
			  popd
			  echo "@ Done! If the backup was successful, you can:" 
			  echo "@ check the contents of the backup by running"
			  echo "  sudo tar -tvf $ZIPFILE"
			  echo "@ restore the backed-up files by running"
			  echo "  sudo tar -xvf $ZIPFILE -C/"
			  echo " @ Note that restoring will overwrite existing files with the same name"
```
- Usage

`sudo /usr/local/src/pwnagotchi/scripts/backup_new.sh testbackup.tar.gz`

- Restore
`sudo tar -xvf /home/pi/testbackup.tar.gz -C/`

cd-rum, 2019-10-20:
Seems worth a PR?

TheVladdestVlad, 2019-10-21:
Yup. I think i might have gotten confused about where the script should be executed. I was mainly trying to create a local backup that i can either extract from the microSD via a card reader, or just grab the backup file using WinSCP (i'm connecting the pwnagotchi to a Windows machine).
Regarding the host-names:
Yes, even the initial host name change brakes stuff since it seems like it doesn't manage to update the /etc/hosts entries for localhost.
The proper steps to restore (at least from what i've tested) would be (considering all the initial set-up steps have been covered):
1. Make sure the host name of your pwnagotchi matches the one in the backup
2. Sync time (if it didn't automatically sync when connected) - tar will error out if it has to restore files that appear to be created in the future
3. Copy the backup to your pwnagotchi unit in /home/pi
4. Run 
`sudo tar -xvf /home/pi/<YourBackupName>.tar.gz -C/`

evilsocket, 2019-10-23:
changed the script

closed 2019-10-23 by evilsocket