evilsocket/pwnagotchi #346: [BUG] pwngrid is still enabled, although disabled in the configuration

strasharo, 2019-10-20:
On a fresh install, although I set the following in /etc/pwnagotchi/config.yml:
```
main:
    plugins:
      grid:
        enabled: false
        report: false

```
I still see that /var/log/pwngrid-peer.log is being written and it's trying to reach the API.
Is the configuration above sufficient to disable pwngrid altogether or the user has also to manually disable pwngrid-peer.service?

**To Reproduce**
Steps to reproduce the behavior:
1. Perform a fresh install
2. Set both options of the grid plugin to false as stated in the documentation.
3. Boot the device
4. Check /var/log/pwngrid-peer.log for signs of current activity
5. Check that pwngrid-peer.service is running and writine the file above

**Expected behavior**
The pwngrid-peer service should honor the configuration and not be active when disabled.

**Environment (please complete the following information):**
 - Pwnagotchi version v1.0.1
 - OS version - Raspbian 10.1 Lite from the release image.
 - Type of hardware RPI0W
 - Any additional hardware used - none

evilsocket, 2019-10-20:
the pwngrid-peer service is not just used for the API but for several other features, that is why it's not disabled even if the grid plugin is. those configuration values determine wether or not the service will be reporting your unit.

closed 2019-10-20 by evilsocket

echox, 2019-11-08:
Is there any way to disable communication to the grid api?
My unit isn't connected to the internet and I will get endless error messages in the log (Post https://api.pwnagotchi.ai/api/v1/unit/enroll:...) I thought disabling the grid plugin might disable this?