evilsocket/pwnagotchi #355: [PROBLEM] pwngrid does not work over TOR

zenzen666, 2019-10-21:
**Describe the bug**
I've tried to use pwngrid using TOR and I got this error:
```err POST https://api.pwnagotchi.ai/api/v1/unit/enroll (10.044264s) Post https://api.pwnagotchi.ai/api/v1/unit/enroll: dial tcp: lookup api.pwnagotchi.ai on 8.8.8.8:53: read udp 10.0.0.2:34188->8.8.8.8:53: i/o timeout
```
I blame cloudflare.

**To Reproduce**
Just try to use pwngrid from a pwnagotchi connected to internet through TOR.

friedphish, 2019-10-22:
This can likely be fixed with a config change on the Cloudflare side to allow Tor without presenting it with the JS challenge/captchas of doom.

Either that or a HS being stood up.

evilsocket, 2019-10-23:
checking the cf side config ;)

evilsocket, 2019-10-31:
@zenzen666 still didn't find a way to disable this part of CF without disabling security alltogether