evilsocket/pwnagotchi #360: Need help to get the Waveshare 2.9 inch e-Paper Display working

cynay, 2019-10-22:
Hi 
I need help to get the very cool pwnagotchi working with the 2.9 Inch e-Paper Display working.
Everything is working except the Display. I can connect over the USB-Ethernet and it's working.
I'm happy for any input :)

Thanks cynay

rooty0, 2019-10-25:
Do you have 
```yaml
ui:
  display:
    type: 'waveshare_2'
    color: 'black'
```

in your `/etc/pwnagotchi/config.yml` ?

cynay, 2019-10-25:
Tried this yes.

I saw that it's necessary to change a few things. I'm trying to get it to work on my device, if everything works i could do a pull request with the additions.

cynay, 2019-10-27:
Hi 
I tried to get it working but because of the code changes I needed to reinstall the pwnagotchi with pip3.
After the reinstall i somehow get this error:

```
Jul 10 02:06:23 pwnagotchi systemd[1]: pwnagotchi.service: Service RestartSec=30s expired, scheduling restart.

Jul 10 02:06:23 pwnagotchi systemd[1]: pwnagotchi.service: Scheduled restart job, restart counter is at 1.

Jul 10 02:06:23 pwnagotchi systemd[1]: Stopped pwnagotchi Deep Reinforcement Learning instrumenting bettercap for WiFI pwning..

Jul 10 02:06:23 pwnagotchi systemd[1]: Started pwnagotchi Deep Reinforcement Learning instrumenting bettercap for WiFI pwning..

Jul 10 02:06:33 pwnagotchi pwnagotchi-launcher[524]: Traceback (most recent call last):

Jul 10 02:06:33 pwnagotchi pwnagotchi-launcher[524]:   File "/usr/local/bin/pwnagotchi", line 4, in <module> 

Jul 10 02:06:33 pwnagotchi pwnagotchi-launcher[524]:     __import__('pkg_resources').require('pwnagotchi==1.1.0rc0')

Jul 10 02:06:33 pwnagotchi pwnagotchi-launcher[524]:   File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 3191, in <module>

Jul 10 02:06:33 pwnagotchi pwnagotchi-launcher[524]:     @_call_aside

Jul 10 02:06:33 pwnagotchi pwnagotchi-launcher[524]:   File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 3175, in _call_aside

Jul 10 02:06:33 pwnagotchi pwnagotchi-launcher[524]:     f(*args, **kwargs)

Jul 10 02:06:33 pwnagotchi pwnagotchi-launcher[524]:   File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 3204, in _initialize_master_working_set

Jul 10 02:06:33 pwnagotchi pwnagotchi-launcher[524]:     working_set = WorkingSet._build_master()

Jul 10 02:06:33 pwnagotchi pwnagotchi-launcher[524]:   File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 583, in _build_master

Jul 10 02:06:33 pwnagotchi pwnagotchi-launcher[524]:     ws.require(__requires__)

Jul 10 02:06:33 pwnagotchi pwnagotchi-launcher[524]:   File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 900, in require

Jul 10 02:06:33 pwnagotchi pwnagotchi-launcher[524]:     needed = self.resolve(parse_requirements(requirements))

Jul 10 02:06:33 pwnagotchi pwnagotchi-launcher[524]:   File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 786, in resolve

Jul 10 02:06:33 pwnagotchi pwnagotchi-launcher[524]:     raise DistributionNotFound(req, requirers)

Jul 10 02:06:33 pwnagotchi pwnagotchi-launcher[524]: pkg_resources.DistributionNotFound: The 'atari-py~=0.2.0; extra == "atari"' distribution was not found and is required by gym

Jul 10 02:06:34 pwnagotchi systemd[1]: pwnagotchi.service: Main process exited, code=exited, status=1/FAILURE

Jul 10 02:06:34 pwnagotchi systemd[1]: pwnagotchi.service: Failed with result 'exit-code'.
```

Could someone help me why suddenly a package is needed which is not found?

Changes I did were adding configs and .py files for the 2.9 Inch screen and adding the option in a few codeblocks where needed.

rooty0, 2019-10-29:
Are you trying this on a clean pwnagotchi install? It works for me out of the box 
[image]

cynay, 2019-10-29:
Yes it's a new install. Is yours the 2.9 Inch display? It looks like another one for me?

rooty0, 2019-10-29:
ah no, sorry, mine is [2.13](https://www.waveshare.com/2.13inch-e-paper-hat.htm), I need more to sleep :(

marsmensch, 2019-10-29:
Please join the slack @cynay. Issues like this can usually be resolved within minutes in the #waveshare channel.

cynay, 2019-10-29:
Wohooo I got it working! It's alive :)

closed 2019-10-29 by cynay

burgerlX, 2019-11-02:
> Wohooo I got it working! It's alive :)

How did you do it? I have one and I need it working

copysolo, 2019-11-28:
Waveshare 2.13 Inch e-Paper Display Hat(B)

 Hi, I have the following problem.  

I've bought this ePaper display on Amazon

 [https://www.amazon.de/gp/product/B075FR81WL/ref=ppx\_yo\_dt\_b\_asin\_title\_o00\_s00?ie=UTF8&psc=1](https://www.amazon.de/gp/product/B075FR81WL/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1) 

, it remains more leery after the installation of pwnagotchi.  I have now tried and found that the configatin waveshare\_2 does not work, I have another display and it only works with the testscript  

https://github.com/waveshare/e-Paper/blob/master/RaspberryPi%26JetsonNano/python/examples/epd\_2in13bc\_test.py  

What do I have to change in the configuration that it works under pwnagotchi.  

Thank you

TheSidd, 2020-06-20:
Hey, how did you make it work on 2.9 display?

cynay, 2020-06-22:
@TheSidd i think it should work out of the box now? I haven't used it in a while :/

Zandiexoder, 2024-01-14:
So just use the waveshare_2 Id?

vishnukubal, 2024-08-10:
> So just use the waveshare_2 Id?

waveshare2in9_v2