evilsocket/pwnagotchi #363: Name Selector on Firstboot

mlinton, 2019-10-22:
It would seem that a good idea would be to set the hostname of each unit randomly on first boot as part of the initialization (like the key material).  This would provide some variety within PwnGrid as well as part of people's own customization.

We could preload the image with the following files:
- https://github.com/danielmiessler/SecLists/tree/master/Usernames/Names

And then do a selection of given and family name from the lists (this might be too westernized a method?)

Then take the two names and concatenate them together to set the hostname in the config.yml file which also makes adjustments to the /etc/hosts /etc/hostname and /etc/motd files.

evilsocket, 2019-10-23:
very westernized ...

evilsocket, 2019-10-23:
i mean, i like the idea, but i don't think those lists would work ...

closed 2019-10-23 by evilsocket