evilsocket/pwnagotchi #367: rotation and acceleration informations for AI

xenDE, 2019-10-23:
what do you think about make the AI better with measuring rotation and acceleration?

let the AI use the information if the pwnie is still on desktop or moving/walking/runnuing/driving?

http://www.raspberrypirobotics.com/measuring-rotation-and-acceleration-with-the-raspberry-pi/

something like this little add on: 

https://www.ebay.de/itm/SV01A103AEA01R00-Board-Mount-Motion-Position-Module-Sensors-Angle-Line-Measuring/133019642719

can this make the AI/deauth better?

evilsocket, 2019-10-23:
the problem is that adding anything to the AI breaks backwards compatibility in terms on brain.nn file :/

evilsocket, 2019-10-23:
ah also something: as part of the observation, the AI looks at how many "misses" it had in the previous loop. that number is basically a side channel for acceleration already

closed 2019-10-23 by evilsocket