evilsocket/pwnagotchi #380: [BUG] Changing the hostname don't update the etc/hosts

shurillu, 2019-10-24:
Hello there, 

   changing the hostname with raspi-config does not update the `/etc/hosts` file, generating the error:

`unable to resolve host <NEW_HOST_NAME>: Temporary failure in name resolution`

### Step to reproduce the error
+log in 
+ `sudo raspi-config`
+ select "Network Options"
+ select "Hostname"
+ give the new hostname name
+ select "Finish"
+ reboot

### Fix
+ log in
+ `sudo nano /etc/hosts`
+ add this line at the end of the file
+ 127.0.0.1 THE_NEW_HOSTNAME_NAME
+ save
+ `sudo reboot`

These steps should be fix the console error message.

Stefano

evilsocket, 2019-10-24:
@caquino

caquino, 2019-10-24:
@evilsocket this was fixed on my last PR, tested with raspi-config.

The issue if that raspi-config/raspbian expected you to change the entry 127.0.1.1 and not 127.0.0.1 like it was changed, moving the change to 127.0.1.1 fixed the issue.

closed 2019-10-24 by evilsocket